Tertiush/bmspace #1: Wiring

gerdluthe, 2022-09-17:
Hello,

can you please leave some details of wiring the PACE bms and which kind of adapter you used to connect from PC to BMS?

Second:
i'd to uncomment the first path und comment out the second one to get the script working

with open(r'config.yaml') as file:
     config = yaml.load(file, Loader=yaml.FullLoader)['options']

#with open(r'/data/options.json') as file:
 #   config = json.load(file)
    print("Config: " + json.dumps(config))

Regards

Tertiush, 2022-09-17:
Thanks for that, I actually fixed it earlier today. The code was originally intended as an HA add-on, and so I missed that.

Wiring depends on your device. I added some notes on mine in the readme.

gerdluthe, 2022-09-17:
Thanks for your reply, that helps me out.
How the config.yaml has to be configured to get an USB/RS232 adapter working?
I tried :
options:
  mqtt_host: "192.168.1.238"
  mqtt_port: 1883
  mqtt_user: ""
  mqtt_password: ""
  mqtt_ha_discovery: false
  mqtt_ha_discovery_topic: "homeassistant"
  mqtt_base_topic: "bmspace"
  connection_type: "Serial"
  bms_ip: "10.0.0.161"
  bms_port: 5000
  bms_serial: "/dev/cu.usbserial-14120"
  scan_interval: 5

gerdluthe, 2022-09-17:
As result:
trying to connect /dev/cu.usbserial-14120
BMS serial connected
BMS Version: 
BMS Serial Number: 
Pack Serial Number: 
Cells: 12
V Cell1: 54028 mV
V Cell2: 55564 mV
V Cell3: 54284 mV
V Cell4: 56076 mV
V Cell5: 53772 mV
V Cell6: 55820 mV
V Cell7: 54796 mV
V Cell8: 55820 mV
V Cell9: 54796 mV
V Cell10: 56076 mV
V Cell11: 55308 mV
V Cell12: 55820 mV
Temperature Sensors: 213
T Cell1: 56.0 Deg
T Cell2: 55.5 Deg
T Cell3: 56.10000000000002 Deg
T Cell4: -119.375 Deg
........

Tertiush, 2022-09-18:
What battery brand & model do you have? I only tested with a Hubble AM2 and AM4.

gerdluthe, 2022-09-18:
I use an no name battery made from an chinese company for me. they used an PACE BMS so i think and also hope that your script would be able to get the data out of the BMS.  the battery is connected to an victron multiplus ii and it would be nice to get them both able to communicate to each other

Tertiush, 2022-09-18:
There's a tool online to monitor this BMS, called PbmsTools V2.5FN

Download it (don't have the link now sorry) and see if you can communicate with it. If you can them PM me a wireshark or similar dump of the comms (from the monitoring screen), maybe I can spot what makes it different.

gerdluthe, 2022-09-18:
OK, i'll search the tool and get back to you. It would took some time.

closed 2023-02-11 by Tertiush