Tertiush/bmspace #3: It is possible to read out a second pack?

cobrav, 2023-01-09:
Hello, 
at first I'd like to thank you for your efforts. It is possible to read out a second pack, connected via RS485 as Slave?
With PBMS-tools I'm able to see the informations provided by the second pack. According to the RS232 protocol it should be possible 
[72651=11625-PACE-RS232commuciation-protocal（PACE-RS232-TY16S）-20180705.pdf](https://github.com/Tertiush/bmspace/files/10372932/72651.11625-PACE-RS232commuciation-protocal.PACE-RS232-TY16S.-20180705.pdf)
Thank you and best regards

Tertiush, 2023-01-09:
Hi, thanks for that document, I actually reverse engineered parts of the
protocol to get to my script!

I also have a second pack so will make work of this. No timeframe
though.... Will hopefully look at it over the weekend

Tertiush, 2023-01-19:
I'm afraid my second pack was borrowed to a friend which had his single one failed. Hopefully back next week then I can get cracking at this.

Tertiush, 2023-02-05:
Update: So things are going pretty good, reading out multiple packs now, should go all the way to 15. I'm looking into implementing some of the other monitoring features, such as alarms / event also before publishing. The mods are quite substantial and will require rework of existing installs, mainly due to renaming unique IDs to allow for multiple packs.

cobrav, 2023-02-05:
Awesome, thanks for the update!

Tertiush, 2023-02-11:
Update released. Please note the breaking changes.

cobrav, 2023-02-11:
Awesome, thanks! Everything is working perfectly. Again thank you for your
efforts.

Tertiush, 2023-02-11:
Awesome, if something pops up let me know

closed 2023-02-11 by Tertiush

cobrav, 2023-09-08:
If you want you can also include my Battery Pack: MES PS5120E to the list. Also working with newer Firmware  P16S100A-0A480-2.21 with multiple batteries (up to 4 tested). Working since February without any issues. Thanks again for your hard work.