Tertiush/bmspace #5: Need a instruction how to start the script with a newstart of the raspberry pi and after an com. failure of the script.

torstenjaeckel, 2023-04-26:
Hi, first i need to thank for the great work of the developper of this script. I run this script with success on windows and a raspberry pi.
However i fail to run the script from a bash file or from cronos or a an instance. How could i create a automatic start up. What is a docker-file if  i run it on a debian linux?

jpmeijers, 2023-04-26:
To address the second part of this issue I am running bms.py in screen, in a while-true loop. This restarts the script whenever the mqtt server disappears, which makes the script crash.

`~/bmspace $ while true; do python bms.py; sleep 10; done;`

I believe the correct way to start and restart the script would be via systemd, but otherwise a docker container (docker compose script) with the restart unless stopped flag should work too.

jpmeijers, 2023-04-26:
I changed my script to run in Docker.

### Installed Docker on RaspberryPi
https://docs.docker.com/engine/install/debian/#install-using-the-convenience-script

`curl -fsSL https://get.docker.com -o get-docker.sh`
`sudo sh get-docker.sh`

https://docs.docker.com/engine/install/linux-postinstall/
`sudo usermod -aG docker ${USER}`
log out, log in
`docker run hello-world`
`sudo systemctl enable docker.service`
`sudo systemctl enable containerd.service`

https://docs.docker.com/config/containers/logging/json-file/
```
{
  "log-driver": "json-file",
  "log-opts": {
    "max-size": "10m",
    "max-file": "3"
  }
}
```

`sudo reboot`

### Build image and run container
`docker compose build`
`docker compose up -d`

### Dockerfile
```
FROM python:3.9-alpine

ENV WORK_DIR=workdir \
  HASSIO_DATA_PATH=/data

RUN mkdir -p ${WORK_DIR}
WORKDIR /${WORK_DIR}
COPY requirements.txt .

# install python libraries
RUN pip3 install -r requirements.txt

# Copy code
COPY bms.py constants.py run.sh ./
RUN chmod a+x run.sh

CMD [ "sh", "./run.sh" ]
```

### docker-compose.yaml
```
version: '3'
services:
  bmspace:
    build:
      context: .
    restart: unless-stopped
    privileged: true
    volumes:
      - ./config.yaml:/workdir/config.yaml
      - /dev:/dev
```

Tertiush, 2023-04-27:
Glad you got it working. Back in the day I ran my scripts as system services using systemd. It also would restart the script if ever failed.

closed 2023-04-27 by Tertiush