Tertiush/bmspace #7: Error parsing BMS warning data

scandiumza, 2023-05-12:
First off, I'd like to thank you for creating and sharing this addon.

I've got a USB to RS232 cable hooked up to one of my two Greenrich U-P5000 batteries and as far as I can tell, the addon, is it's pretty much correctly parsing everything correctly except for the warnings. 

I've attached the complete (and verbose) log output leading up to the parsing error. Hope this helps :)

[bmspace.log](https://github.com/Tertiush/bmspace/files/11465611/bmspace.log)

Tertiush, 2023-05-12:
Found the issue, fix should come tomorrow. Problem was the protocol sometimes adds a INFOFLAG in the byte-stream which shifts the data out. I've added logic to test for this.

Will also try to reduce the length of reported warning info as it sometimes hits HA's character limit.

Tertiush, 2023-05-13:
Update pushed. Please test and let me know of there's still issues. thanks

scandiumza, 2023-05-13:
Hi Tertius, 

Thanks for getting this sorted out so quickly, I've updated and can confirm that a) the errors are gone and b) I'm now able to see the warnings in HA. This issue looks to be resolved, so I'll be closing this issue 😊

closed 2023-05-13 by scandiumza