Tertiush/bmspace #11: Connection to Pace BMS via Elfin Tcp-rs232

snaven1, 2023-05-21:
I own a Jakiper Model Jk48V100 Pro Battery Pack that is running a Pace BMS. Can access the BMS via the RS232 Port using PBMStools displaying all data, without issue. This uses a serial rs232 to usb cable connection. Have installed the Home assistant BMSpace software without issue. Trying to connect to the BMS via tcp through an Elfin-EE10A TCP-RS232232 convertor. Checked all wiring very carefully, so do not believe this is an issue. Have used these elfin devices previously without issue and seem to work very well.
Also, somewhat confused as to what bms_port number, bms_serial interface should be used (if at all) given that it a tcp connection.

Output error from Home Assistant ...

Hello BMS Pace
Starting up...
Loading options.json
Config: {"mqtt_host": "192.168.100.86", "mqtt_port": 1883, "mqtt_user": "mqtt", "mqtt_password": "mqtt", "mqtt_ha_discovery": true, "mqtt_ha_discovery_topic": "homeassistant", "mqtt_base_topic": "bmspace", "connection_type": "IP", "bms_ip": "192.168.100.120", "bms_port": 5000, "bms_serial": "/dev/ttyUSB0", "scan_interval": 5, "debug_output": 2}
Connection Type: IP
MQTT connected with result code 0
Connecting to BMS...
trying to connect 192.168.100.120:5000
BMS socket error connecting: [Errno 111] Connection refused
BMS socket error: 'bool' object has no attribute 'send'
Error, connection to BMS lost
Error retrieving BMS version number
BMS socket error: 'bool' object has no attribute 'send'
Error, connection to BMS lost
Error: Error, connection to BMS lost
Traceback (most recent call last):
  File "/workdir/./bms.py", line 1097, in <module>
    pack_sn.replace(" ", "")
AttributeError: 'bool' object has no attribute 'replace'
Script exiting

Any suggestions would be greatly appreciated ...

Tertiush, 2023-05-21:
The script is being refused the connection to the socket (IP:port). Likely another app/service is using the socket (if only one connection is allowed), the IP or port is incorrect or a firewall might be blocking it. Can't think of anything else now

The serial settings are ignored if IP is selected.

Tertiush, 2023-05-21:
It could also be that the TCP-Serial convertor is still expecting another host to connect from a previous session, i.e. it hasn't yet timed-out... Have you tried restarting / power cycling the convertor?

snaven1, 2023-05-21:
Thanks for the quick response, have rebooted elfin device without any further success. Will research a little further, need to test the wiring to absolutely ensure that everything is wired correctly. There are also a number of options within the elfin that can be changed. Has a built in web site. Thanks for the assist. As a matter of interest, I have used the rs485 port on the Jakiper battery and extracted modbus data via an elfin ee11 rs485 to tcp conveter with success.

snaven1, 2023-05-22:
Have done further research, have absolutely ruled out any wiring issues. I noticed the Pace RS232 manual mentioned that it runs in serial frame mode, I actually turned this on within the elfin device without success. I was just wondering what did you use to test the ip-tcp connection method when building your code structure. Furthermore, I am still dubious about the error messages being presented by the log output mentioned above, especially about the connection message  "trying to connect 192.168.100.120:5000". Any further thoughts would be greatly appreciated.
Thanks

Tertiush, 2023-05-23:
I also use a TCP-serial converter, but I don't know the brand offhand. I
believe most users use the IP connection type.

"trying to connect 192.168.100.120:5000" means it's trying to connect to IP
192.168.100.120 on TCP port 5000. This is what you configured as the
TCP-serial converter's address.

snaven1, 2023-05-24:
Fully operational, Port no. issue. The TCP server integrated into the elfin device was misconfigured, needed to have  "Tcp- Server" setup with a port number that is used within the Home assistant configuration file. Can confirm that a RS232 -Tcp Elfin EE10 device works with a Jakiper Model Jk48V100 Pro Battery Pack without issue. Additionally, I am running two battery packs in parallel, and all data being extracted correctly! Additionally, should be noted that Jakiper batery is identical to the SOK battery, so the same combination should work without issue.

Home Assistant log output ....

SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'B050'
 - LCHKSUM:  66
 - LENID:  b'050'
 - LENID (int):  80
INFO:  b'323135313931303233393030303930502020202053657020203320323032322C31363A33353A3131'
CHKSUM:  b'ED9B'
Calc CHKSUM:  ED9B
BMS Serial Number: 215191023900090
Pack Serial Number: Sep  3 2022,16
-> Outgoing Data:  b'~25014642E002FFFD05\r'
<- Incoming data:  b'~25014600D0F40002100CDB0CDB0CDB0CDB0CDB0CDC0CDB0CDB0CDB0CDC0CDC0CDC0CDB0CDB0CDB0CDB060B7A0B7D0B790B7B0B8C0B850000CDB4114E0227DE000127102B100CE30CE20CE30CE30CE30CE30CE30CE30CE30CE30CE30CE30CE30CE30CE30CE1060B700B6E0B6C0B6D0B770B750000CE2D14E30228B50000271033C66B\r'
SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'D0F4'
 - LCHKSUM:  68
 - LENID:  b'0F4'
 - LENID (int):  244
INFO:  b'0002100CDB0CDB0CDB0CDB0CDB0CDC0CDB0CDB0CDB0CDC0CDC0CDC0CDB0CDB0CDB0CDB060B7A0B7D0B790B7B0B8C0B850000CDB4114E0227DE000127102B100CE30CE20CE30CE30CE30CE30CE30CE30CE30CE30CE30CE30CE30CE30CE30CE1060B700B6E0B6C0B6D0B770B750000CE2D14E30228B50000271033'
CHKSUM:  b'C66B'
Calc CHKSUM:  C66B
Packs: 2
Pack 1, Total cells: 16
Pack 1, V Cell1: 3291 mV
Pack 1, V Cell2: 3291 mV
Pack 1, V Cell3: 3291 mV
Pack 1, V Cell4: 3291 mV
Pack 1, V Cell5: 3291 mV
Pack 1, V Cell6: 3292 mV
Pack 1, V Cell7: 3291 mV
Pack 1, V Cell8: 3291 mV
Pack 1, V Cell9: 3291 mV
Pack 1, V Cell10: 3292 mV
Pack 1, V Cell11: 3292 mV
Pack 1, V Cell12: 3292 mV
Pack 1, V Cell13: 3291 mV
Pack 1, V Cell14: 3291 mV
Pack 1, V Cell15: 3291 mV
Pack 1, V Cell16: 3291 mV
Pack 1, Cell Max Diff Volt Calc: 1 mV
Pack 1, Total temperature sensors: 6
Pack 1, Temp1: 20.8 ℃
Pack 1, Temp2: 21.1 ℃
Pack 1, Temp3: 20.7 ℃
Pack 1, Temp4: 20.9 ℃
Pack 1, Temp5: 22.6 ℃
Pack 1, Temp6: 21.9 ℃
Pack 1, I Pack: 0.0 A
Pack 1, V Pack: 52.66 V
Pack 1, I Remaining Capacity: 44300 mAh
Pack 1, I Full Capacity: 102060 mAh
Pack 1, SOC: 43.41 %
Pack 1, Cycles: 1
Pack 1, Design Capacity: 100000 mAh
Pack 1, SOH: 102.06 %
Pack 2, Total cells: 16
Pack 2, V Cell1: 3299 mV
Pack 2, V Cell2: 3298 mV
Pack 2, V Cell3: 3299 mV
Pack 2, V Cell4: 3299 mV
Pack 2, V Cell5: 3299 mV
Pack 2, V Cell6: 3299 mV
Pack 2, V Cell7: 3299 mV
Pack 2, V Cell8: 3299 mV
Pack 2, V Cell9: 3299 mV
Pack 2, V Cell10: 3299 mV
Pack 2, V Cell11: 3299 mV
Pack 2, V Cell12: 3299 mV
Pack 2, V Cell13: 3299 mV
Pack 2, V Cell14: 3299 mV
Pack 2, V Cell15: 3299 mV
Pack 2, V Cell16: 3297 mV
Pack 2, Cell Max Diff Volt Calc: 2 mV
Pack 2, Total temperature sensors: 6
Pack 2, Temp1: 19.8 ℃
Pack 2, Temp2: 19.6 ℃
Pack 2, Temp3: 19.4 ℃
Pack 2, Temp4: 19.5 ℃
Pack 2, Temp5: 20.5 ℃
Pack 2, Temp6: 20.3 ℃
Pack 2, I Pack: 0.0 A
Pack 2, V Pack: 52.781 V
Pack 2, I Remaining Capacity: 53470 mAh
Pack 2, I Full Capacity: 104210 mAh
Pack 2, SOC: 51.31 %
Pack 2, Cycles: 0
Pack 2, Design Capacity: 100000 mAh
Pack 2, SOH: 104.21 %
Script running....

closed 2023-05-24 by snaven1