Tertiush/bmspace #13: Cannot get data out of BMS most of the time

disaster123, 2023-05-25:
While the original Pace BMS Tool works fine to read all data under windows. I don't get data with bmspace or most of the time. Sometimes it works for a single command.

To speed up testing i removed the mqtt part and added just two debug prints after outgoing data (binary data and length)

Example:
```
tarting up...
Loading config.yaml
Connection Type: Serial
Connecting to BMS...
trying to connect /dev/ttyUSB1
BMS serial connected
-> Outgoing Data:  b'~250146C10000FD9A\r'
0
b''
```

Working example for a single command - after trying 10-20 times:
```
Starting up...
Loading config.yaml
Connection Type: Serial
Connecting to BMS...
trying to connect /dev/ttyUSB1
BMS serial connected
-> Outgoing Data:  b'~250146C10000FD9A\r'
58
b'~25014600602850313653323030412D32313338322D322E303200F585\r'
<- Incoming data:  b'~25014600602850313653323030412D32313338322D322E303200F585\r'
SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'6028'
 - LCHKSUM:  54
 - LENID:  b'028'
 - LENID (int):  40
INFO:  b'50313653323030412D32313338322D322E303200'
CHKSUM:  b'F585'
Calc CHKSUM:  F585
BMS Version: P16S200A-21382-2.02
-> Outgoing Data:  b'~250146C20000FD99\r'
0
b''

```

Tertiush, 2023-05-26:
I need more info, and more logs to see if there's a pattern.

disaster123, 2023-05-26:
I found out it‘s only not working on a raspberry pi - works fine with my huawai notebook :-( any ideas?

disaster123, 2023-05-26:
oh man - got it fixed - the OS i'm using on the raspberry pi automatically attached another process to the device trying to detect hardware on that port.... Thanks for your great software!

closed 2023-05-26 by disaster123