Tertiush/bmspace #15: Config.yaml error

kebaotu, 2023-05-31:
I use a computer A run ubuntu and I connect usb ch340 from my computer to bms (rj11). I connect computer A to my network. I have another computer run home assistant with mqtt broker. How can I config config.yaml on my computer A for send infomation of my batterie to my home assistant?
My batterie use bms pace (I can read with bpms tools)
Thank you

Tertiush, 2023-05-31:
What is the exact error the script is giving? Paste the log output here.
Without this I cannot help you

kebaotu, 2023-06-01:
my config.ya
options:
  mqtt_host: "IP homeassistant"
  mqtt_port: 1883
  mqtt_user: "users"
  mqtt_password: "password"
  mqtt_ha_discovery: true
  mqtt_ha_discovery_topic: "homeassistant"
  mqtt_base_topic: "bmspace"
  connection_type: "Serial"
  bms_ip: "IP computer run bms.py"
  bms_port: 5000
  bms_serial: "/dev/ttyUSB0"
  scan_interval: 5
  debug_output: 0
- Computer ubuntu A output:
starting up...
Loading Type: serial
MQTT connected with result code 0
connecting to BMS..
trying to connect /dev/ttyUSB0
BMS serial connected
BMS version: PS16100A-....
Pack Serial Number: S0079GT3L...
Pack 1.....
Pack 2...
- Computer B run home assistant:
Mosquitto broker log:
2023-06-01 12:40:48: New connection from 172.30.32.2:39722 on port 1883.
2023-06-01 12:40:48: Client <unknown> closed its connection.
2023-06-01 12:42:48: New connection from 172.30.32.2:48966 on port 1883.
2023-06-01 12:42:48: Client <unknown> closed its connection.
2023-06-01 12:44:48: New connection from 172.30.32.2:50488 on port 1883.
2023-06-01 12:44:48: Client <unknown> closed its connection....

In core-mosquitto, I can't find any device

Tertiush, 2023-06-01:
>    2023-06-01 12:40:48: Client closed its connection.
>    2023-06-01 12:42:48: Client closed its connection.
>    2023-06-01 12:44:48: Client closed its connection....

Please change the setting '*debug_output'* to *3* and send that log output.

Tertiush, 2023-06-02:
There's two things related to the MQTT connection I can see from your log. Although the MQTT connection is initially made, it soon disconnects with error code 7. This could be a:

1. network issue or,
2. according to a google search another client might be connecting using the same clientID. This is however unlikely as this client generates a random ID when initialised.
 
So its likely a network issue, or just your broker someone forcing the disconnect and preventing a re-connect

kebaotu, 2023-06-03:
I had to add the sensor manually in configuration
Ex
`mqtt:
  sensor:
  - name: "Battery soc"
    state_topic: "bmspace/pack_1/soc"
    unit_of_measurement: "%"
  - name: "Battery voltage"
    state_topic: "bmspace/pack_1/v_pack"
    unit_of_measurement: "v"
  - name: "Battery current"
    state_topic: "bmspace/pack_1/i_pack"
    unit_of_measurement: "A"
  - name: "Battery temperature 6"
    state_topic: "bmspace/pack_1/temps/temp_6"
    unit_of_measurement: "C"`

kebaotu, 2023-06-03:
>

closed 2023-06-03 by kebaotu