Tertiush/bmspace #17: Checksum error

RoganDawes, 2023-06-20:
Hi, I am seeing checksum errors trying to monitor my two Greenrich U-P5000 5kWh battery packs.

```
python3 ./bms.py
Starting up...
Loading config.yaml
Connection Type: IP
MQTT connected with result code 0
Connecting to BMS...
trying to connect 192.168.201.1:5000
BMS socket connected
BMS Version: P16S150A-17900-2.01
BMS Serial Number: 17900112250937D
Pack Serial Number: Jan102022,13
Error retrieving BMS analog data: Checksum error
Pack Remaining Capacity: 101380 mAh
Pack Full Capacity: 106270 mAh
Pack Design Capacity: 99000 mAh
Pack SOC: 95.4 %
Pack SOH: 107.34 %
Packs for warnings: 2
Pack 1, warnings:
Pack 1, balancing1: 00000000
Pack 1, balancing2: 00000000
Pack 2, warnings:
Pack 2, balancing1: 00000000
Pack 2, balancing2: 00000000
Publishing HA Discovery topic...
Error retrieving BMS analog data: Checksum error
Error retrieving BMS analog data: Checksum error
Error retrieving BMS analog data: Checksum error
Error retrieving BMS analog data: Checksum error
Error retrieving BMS analog data: Checksum error
Error retrieving BMS analog data: Checksum error
Error retrieving BMS analog data: Checksum error
Error retrieving BMS warning info: Checksum error
Error retrieving BMS analog data: Checksum error
Error retrieving BMS analog data: Checksum error
```

I am connected via RS232 to the pack, but running bms.py on my M1 Mac through ser2net on Linux to reach the actual ttyUSB5 device. The two packs are connected together via RS485, and to the inverter via CAN.

A snippet of the log run with debug level 3:
```
Received and calculated CHKSUM does not match: Received: , Calculated: D7DD
...for incoming data: b'~25014600D0F40002100D360D370D380D350D370D370D390D370D390D390D380D370D370D380D380D37060B820B7F0B7F0B860B780B8C04F3D3772730022983010526AC5E100D380D3A0D3A0D380D390D380D3A0D390D3C0D3B0D3B0D390D3\r' |Hex: 7e 32 35 30 31 34 36 30 30 44 30 46 34 30 30 30 32 31 30 30 44 33 36 30 44 33 37 30 44 33 38 30 44 33 35 30 44 33 37 30 44 33 37 30 44 33 39 30 44 33 37 30 44 33 39 30 44 33 39 30 44 33 38 30 44 33 37 30 44 33 37 30 44 33 38 30 44 33 38 30 44 33 37 30 36 30 42 38 32 30 42 37 46 30 42 37 46 30 42 38 36 30 42 37 38 30 42 38 43 30 34 46 33 44 33 37 37 32 37 33 30 30 32 32 39 38 33 30 31 30 35 32 36 41 43 35 45 31 30 30 44 33 38 30 44 33 41 30 44 33 41 30 44 33 38 30 44 33 39 30 44 33 38 30 44 33 41 30 44 33 39 30 44 33 43 30 44 33 42 30 44 33 42 30 44 33 39 30 44 33 0d
Length of incoming data as measured: 191
SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN (decode!):  b'00'
LENGTH:  b'D0F4'
 - LCHKSUM:  68
 - LENID:  b'0F4'
 - LENID (int):  244
```

Please let me know if there is any more information I can provide?

Tertiush, 2023-06-21:
I also get quite a few errors on my line, guess in part to my 7+ metre long
serial link.... Do you still get data updates regularly / on most poll
intervals?

RoganDawes, 2023-06-21:
No,I get zero readings after the initial serial number, etc shown. Only checksum errors.

Tertiush, 2023-06-21:
Ok, I'll have a look over the weekend. It will help having a longer debug
trace, covering a few of the failures.

RoganDawes, 2023-06-21:
It seems to be connected to the second battery. Mine died while the vendor was trying to upgrade the firmware, and now only a single battery is replying, and the checksums are fine.
I do however still have to take my batteries to the vendor to sort themselves out.

RoganDawes, 2023-06-29:
I'll reopen this if it is still present once I have the batteries sorted out.

RoganDawes, 2023-07-01:
Got my batteries back online, and confirmed that I am still getting the checksum error:
```
python3 ./bms.py
Starting up...
Loading config.yaml
Connection Type: IP
MQTT connected with result code 0
Connecting to BMS...
trying to connect 192.168.201.5:5000
BMS socket connected
-> Outgoing Data:  b'~250146C10000FD9A\r'
<- Incoming data:  b'~25014600602850313653313530412D31373930302D322E303557F571\r'
SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'6028'
 - LCHKSUM:  54
 - LENID:  b'028'
 - LENID (int):  40
INFO:  b'50313653313530412D31373930302D322E303557'
CHKSUM:  b'F571'
Calc CHKSUM:  F571
BMS Version: P16S150A-17900-2.05W
-> Outgoing Data:  b'~250146C20000FD99\r'
<- Incoming data:  b'~25014600B05031373930303131323235303933374420202020204A616E20313020323032322C31333A34343A3133ED79\r'
SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'B050'
 - LCHKSUM:  66
 - LENID:  b'050'
 - LENID (int):  80
INFO:  b'31373930303131323235303933374420202020204A616E20313020323032322C31333A34343A3133'
CHKSUM:  b'ED79'
Calc CHKSUM:  ED79
BMS Serial Number: 17900112250937D
Pack Serial Number: Jan102022,13
-> Outgoing Data:  b'~25014642E002FFFD05\r'
<- Incoming data:  b'~25014600D0F40002100DD60DBC0DD70DD70DD40DD70DD20DD60DD30DD60DC10DD40DD50DD70DD30DD5060B760B710B700B7A0B7D0B9D0000DD2526A90226AC011126AC64100DD30DBD0DD40DC60DD50DD40DD50DD50DD60DD60DD40DD20DD30\r'
SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'D0F4'
 - LCHKSUM:  68
 - LENID:  b'0F4'
 - LENID (int):  244
INFO:  b'0002100DD60DBC0DD70DD70DD40DD70DD20DD60DD30DD60DC10DD40DD50DD70DD30DD5060B760B710B700B7A0B7D0B9D0000DD2526A90226AC011126AC64100DD30DBD0DD40DC60DD50DD40DD50DD50DD60DD60DD40DD20DD30\r'
CHKSUM:  b''
Calc CHKSUM:  D61A
Received and calculated CHKSUM does not match: Received: , Calculated: D61A
...for incoming data: b'~25014600D0F40002100DD60DBC0DD70DD70DD40DD70DD20DD60DD30DD60DC10DD40DD50DD70DD30DD5060B760B710B700B7A0B7D0B9D0000DD2526A90226AC011126AC64100DD30DBD0DD40DC60DD50DD40DD50DD50DD60DD60DD40DD20DD30\r' |Hex: 7e 32 35 30 31 34 36 30 30 44 30 46 34 30 30 30 32 31 30 30 44 44 36 30 44 42 43 30 44 44 37 30 44 44 37 30 44 44 34 30 44 44 37 30 44 44 32 30 44 44 36 30 44 44 33 30 44 44 36 30 44 43 31 30 44 44 34 30 44 44 35 30 44 44 37 30 44 44 33 30 44 44 35 30 36 30 42 37 36 30 42 37 31 30 42 37 30 30 42 37 41 30 42 37 44 30 42 39 44 30 30 30 30 44 44 32 35 32 36 41 39 30 32 32 36 41 43 30 31 31 31 32 36 41 43 36 34 31 30 30 44 44 33 30 44 42 44 30 44 44 34 30 44 43 36 30 44 44 35 30 44 44 34 30 44 44 35 30 44 44 35 30 44 44 36 30 44 44 36 30 44 44 34 30 44 44 32 30 44 44 33 30 0d
Length of incoming data as measured: 193
SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN (decode!):  b'00'
LENGTH:  b'D0F4'
 - LCHKSUM:  68
 - LENID:  b'0F4'
 - LENID (int):  244
INFO:  b'0002100DD60DBC0DD70DD70DD40DD70DD20DD60DD30DD60DC10DD40DD50DD70DD30DD5060B760B710B700B7A0B7D0B9D0000DD2526A90226AC011126AC64100DD30DBD0DD40DC60DD50DD40DD50DD50DD60DD60DD40DD20DD30\r'
CHKSUM:  b''
Error retrieving BMS analog data: Checksum error
[... 245 lines cut ...]
 - LENID (int):  244
INFO:  b'0002100DD50DBC0DD70DD70DD40DD60DD20DD50DD30DD60DC10DD40DD50DD70DD30DD5060B760B710B700B7A0B7D0B9D0000DD2226A90226AC011126AC64100DD30DBC0DD40DC60DD50DD40DD50DD50DD60DD60DD40DD20DD30\r'
CHKSUM:  b''
Calc CHKSUM:  D621
Received and calculated CHKSUM does not match: Received: , Calculated: D621
...for incoming data: b'~25014600D0F40002100DD50DBC0DD70DD70DD40DD60DD20DD50DD30DD60DC10DD40DD50DD70DD30DD5060B760B710B700B7A0B7D0B9D0000DD2226A90226AC011126AC64100DD30DBC0DD40DC60DD50DD40DD50DD50DD60DD60DD40DD20DD30\r' |Hex: 7e 32 35 30 31 34 36 30 30 44 30 46 34 30 30 30 32 31 30 30 44 44 35 30 44 42 43 30 44 44 37 30 44 44 37 30 44 44 34 30 44 44 36 30 44 44 32 30 44 44 35 30 44 44 33 30 44 44 36 30 44 43 31 30 44 44 34 30 44 44 35 30 44 44 37 30 44 44 33 30 44 44 35 30 36 30 42 37 36 30 42 37 31 30 42 37 30 30 42 37 41 30 42 37 44 30 42 39 44 30 30 30 30 44 44 32 32 32 36 41 39 30 32 32 36 41 43 30 31 31 31 32 36 41 43 36 34 31 30 30 44 44 33 30 44 42 43 30 44 44 34 30 44 43 36 30 44 44 35 30 44 44 34 30 44 44 35 30 44 44 35 30 44 44 36 30 44 44 36 30 44 44 34 30 44 44 32 30 44 44 33 30 0d
Length of incoming data as measured: 193
SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN (decode!):  b'00'
LENGTH:  b'D0F4'
 - LCHKSUM:  68
 - LENID:  b'0F4'
 - LENID (int):  244
INFO:  b'0002100DD50DBC0DD70DD70DD40DD60DD20DD50DD30DD60DC10DD40DD50DD70DD30DD5060B760B710B700B7A0B7D0B9D0000DD2226A90226AC011126AC64100DD30DBC0DD40DC60DD50DD40DD50DD50DD60DD60DD40DD20DD30\r'
CHKSUM:  b''
Error retrieving BMS analog data: Checksum error
^CTraceback (most recent call last):
  File "/Users/rogan/workspace/bmspace/./bms.py", line 1117, in <module>
    time.sleep(scan_interval/3)
KeyboardInterrupt
Script exiting
```

Tertiush, 2023-07-01:
Ok will have a look, but cannot soon due to family matters....

RoganDawes, 2023-07-01:
> Ok will have a look, but cannot soon due to family matters.... 

Whenever you get a chance. No rush at all.

RoganDawes, 2023-07-04:
for what it is worth, changing batNumber from 255 to 1 stops the errors
```
success, data = bms_getAnalogData(bms,batNumber=1)
```
But obviously doesn't do anything for the second pack!

Tertiush, 2023-07-07:
Quick check seems to suggest there's been a protocol change. I see your version is 2.05W, whereas mine is 1.46T.

When reading multiple packs the packing of data appears different, as the LENID parameter, indicating the length of data doesn't match the actual length. My only conclusion atm is a change in the protocol. I'll have another look and unpack one of the streams next to the protocol datasheet I have to check if this is likely the case. Hopefully sometime over the weekend.....

RoganDawes, 2023-07-07:
Much appreciated!

Tertiush, 2023-07-08:
Right, so It seems like the data returned is actually correct, however it always cuts off longer replies after 192 bytes (which is the case for analog & warning info). This is why the checksum is failing. There's likely some setting in your serial drivers that's causing this.

RoganDawes, 2023-07-08:
Interesting! I'm going via ser2net, so the socket read is obviously shorter and you don't know to read a second time. Will see if there is a setting I can change.

Tertiush, 2023-07-08:
Well technically I can read a second time, but that would mean the code will have to expect more. Currently its not setup like that any rely on the underlaying drivers / port servers to read until a timeout or EOL is detected.

RoganDawes, 2023-07-09:
Well, you nailed it, I ran it with direct access to the serial port instead of via ser2net, and it works perfectly. Now to see how I can configure ser2net to read more by default. I'm sure I remember something about a read timeout in 10ths of a character period, but I cannot find it now. Will keep digging, and will update here if I find it!

Thanks for the assistance! It is much appreciated!

RoganDawes, 2023-07-09:
So, ser2net has the chardelay* parameters, but I was unable to change the behaviour of the program using them. However, I re-read your bms_get_data routine and realised WHY you have the logic to "decide which one to take":

```
def bms_get_data(comms):
    try:
        if connection_type == "Serial":
            inc_data = comms.readline()
        else:
            temp = comms.recv(4096)
            temp2 = temp.split(b'\r')
            # Decide which one to take:
            for element in range(0,len(temp2)):
                SOI = hex(ord(temp2[element][0:1]))
                if SOI == '0x7e':
                    inc_data = temp2[element] + b'\r'
                    break
```

It's because occasionally the socket read returns less than the full line, and the rest is dangling in the TCP buffer for the next read. When you try to read for the next command, you read the dangling data, as well as the new response, and have to try to "resync".

My thinking is to make it more robust to read the full length of the line in the first place, by either checking if the data read ends with the EOI character (simple), or by parsing out the expected length of data, and ensuring that we have that much data, plus the expected number of trailing characters (a little more complicated, but not much, and probably a little more reliable than the first approach).

This change below makes it work perfectly for me:
```
def bms_get_data(comms):
    try:
        if connection_type == "Serial":
            inc_data = comms.readline()
        else:
            temp = bytes()
            while len(temp) == 0 or temp[-1] != 13:
                temp = temp + comms.recv(4096)
```

RoganDawes, 2023-07-09:
> My thinking is to make it more robust to read the full length of the line in the first place, by either checking if the data read ends with the EOI character (simple), or by parsing out the expected [...]

The problem with the simple approach is that it might coincidentally terminate the read on a 0x13 data byte in the middle of the line. By parsing out the expected data length, we can be absolutely sure we have all the data.

I'm honestly not sure how this hasn't bitten you on the direct serial port side of things, unless the protocol is specifically designed to never include that character in the body? Seems unlikely given all the checksums and length values that could have a 0x13 in them!

Tertiush, 2023-07-09:
Agree with your analysis, thanks: Check for the start sequence, extract the LENID, then use it to ensure the full reply is read in. Seems like a job for next weekend ;)

RoganDawes, 2023-07-09:
Sorry, I obviously meant 0x0D, not 0x13!

I started by copying bms_parse_data to bms_parse_len, but also began converting to raising Exceptions instead of returning False to indicate an error. Deleted it though when I realised the simple approach might work 😂 Still think it's a good solution long term though.

Then bms_parse_data can skip the first few fields already validated, and simply validate the final checksums?

Tertiush, 2023-08-11:
Closing for now, at some time will merge the interim fix to prod version. Thanks for the help

closed 2023-08-11 by Tertiush