Tertiush/bmspace #19: no client id

disaster123, 2023-07-17:
i'm running iobroker with bmspace but the mqtt server always reports - no client connected even tough bmspace connects fine. The  issue is that bmspace sends no client id.

Log:
```
Client [] connected with secret 1689627831142_9164
```

Normally in [] is the mqtt client id written.

Greets,
Stefan

Tertiush, 2023-07-18:
Weird indeed, as the library utilised (paho_mqtt) generates an automatic client id. No-one else has reported this as being an issue, so will need more debug logs from your instance to see what the issue is. I doubt its the paho_mqtt library.

disaster123, 2023-07-18:
To me paho_mqtt does not automatically generate a client id.

This is fixed by:
```
@ -66,7 +69,7 @@ def on_disconnect(client, userdata, rc):
     mqtt_connected = False
 
 
-client = mqtt.Client()
+client = mqtt.Client("bmspace")
 client.on_connect = on_connect
 client.on_disconnect = on_disconnect
 #client.on_message = on_message

```

disaster123, 2023-09-02:
This is still valid

Tertiush, 2023-09-02:
I've forced the client id in the dev version. Please use that version for now, until its pushed to the main.

closed 2023-09-02 by Tertiush