Tertiush/bmspace #20: Compatible with Youthpower pack

jean-luc1203, 2023-07-21:
Hello, this message is to inform you that your code works with batteries:
[YouthPower](https://www.youth-power.net/products/solar-energy-storage-batteries) Rack Module 48V 100AH 4U-5U Lifepo4 Solar Storage Battery

Do you think it is possible to interrogate a 2nd battery connected to the 1st via your code?

Tertiush, 2023-07-21:
Great stuff thanks. As long as you are plugged into the master battery, any other batteries in the same pack (using the interconnected cable) should be picked up automatically. Is this your setup?

jean-luc1203, 2023-07-21:
The equipment isn't at my place, it's at a friend's 700 km away. I've just contacted him to ask him to take a photo of his 2 batteries to check that it's connected to the 1st one and that there's a communication cable between the 2 batteries. He'll get back to me in 2 hours.
In the meantime, here's what I got from the YouhtPower topic.
Does it help?

`pi@Onduleur:~/YouthPower$ mosquitto_sub -h 127.0.0.1 -t YouthPower/# -v
YouthPower/v_cells/cell_1 3314
YouthPower/v_cells/cell_2 3315
YouthPower/v_cells/cell_3 3314
YouthPower/v_cells/cell_4 3314
YouthPower/v_cells/cell_5 3315
YouthPower/v_cells/cell_6 3315
YouthPower/v_cells/cell_7 3315
YouthPower/v_cells/cell_8 3314
YouthPower/v_cells/cell_9 3315
YouthPower/v_cells/cell_10 3314
YouthPower/v_cells/cell_11 3315
YouthPower/v_cells/cell_12 3314
YouthPower/v_cells/cell_13 3315
YouthPower/v_cells/cell_14 3314
YouthPower/v_cells/cell_15 3314
YouthPower/t_cells/cell_1 28.1
YouthPower/t_cells/cell_2 28.3
YouthPower/t_cells/cell_3 27.8
YouthPower/t_cells/cell_4 27.9
YouthPower/t_mos 28.8
YouthPower/t_env 29.9
YouthPower/pack/i_pack -4.77
YouthPower/pack/v_pack 49.766
YouthPower/pack/i_cap 94940
YouthPower/pack/soh 102.94
YouthPower/pack/cycles 290
YouthPower/pack/i_full 100000
YouthPower/pack/soc 92
YouthPower/v_cells/cell_1 3314
YouthPower/v_cells/cell_2 3315
YouthPower/v_cells/cell_3 3314
YouthPower/v_cells/cell_4 3314
YouthPower/v_cells/cell_5 3315
YouthPower/v_cells/cell_6 3315
YouthPower/v_cells/cell_7 3315
YouthPower/v_cells/cell_8 3314
YouthPower/v_cells/cell_9 3315
YouthPower/v_cells/cell_10 3314
YouthPower/v_cells/cell_11 3315
YouthPower/v_cells/cell_12 3314
YouthPower/v_cells/cell_13 3314
YouthPower/v_cells/cell_14 3314
YouthPower/v_cells/cell_15 3314
YouthPower/t_cells/cell_1 28.1
YouthPower/t_cells/cell_2 28.3
YouthPower/t_cells/cell_3 27.8
YouthPower/t_cells/cell_4 27.9
YouthPower/t_mos 28.8
YouthPower/t_env 29.9
YouthPower/pack/i_pack -4.87
YouthPower/pack/v_pack 49.765
YouthPower/pack/i_cap 94940
YouthPower/pack/soh 102.94
YouthPower/pack/cycles 290
YouthPower/pack/i_full 100000
YouthPower/pack/soc 92
`
Here's what the attachment looks like with the current topics

[image]

Tertiush, 2023-07-21:
Are you sure the latest script version is running? Just confirm that as I
don't see "pack_1" as part of the topic, which was introduced when multiple
packs got supported.

jean-luc1203, 2023-07-21:
[image]

here they are, there's an rs485 connection cable between the 2.
In any case, thanks for the answers

jean-luc1203, 2023-07-21:
You're right. I didn't take the latest version, the one I have is older. Sorry I forgot to update. I'll do it tomorrow at the end of the afternoon and come back to give you some feedback.

jean-luc1203, 2023-07-22:
Thank you very much, your code works like a charm. I was able to integrate it easily. 
My friend is very happy with your work and mine.

I quoted the source of your code on my forum. It's in French, so if you'd like to have a look, click [here](http://domosimple.eu/forum/thread-280-post-2166.html#pid2166).

[image]

[image]

it's great

Tertiush, 2023-07-22:
That's great news, thanks for sharing!

closed 2023-07-22 by Tertiush