Tertiush/bmspace #22: Reading 2nd pack does not work

tuxedo0801, 2023-07-30:
Reading 2nd pack does not work with latest code.

```
Starting up...
Loading config.yaml
Connection Type: Serial
MQTT connected with result code 0
Connecting to BMS...
trying to connect /dev/ttyUSB0
BMS serial connected
SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'6028'
 - LCHKSUM:  54
 - LENID:  b'028'
 - LENID (int):  40
INFO:  b'50313653323030412D32313630362D312E303100'
CHKSUM:  b'F588'
Calc CHKSUM:  F588
BMS Version: P16S200A-21606-1.01@
SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'B050'
 - LCHKSUM:  66
 - LENID:  b'050'
 - LENID (int):  80
INFO:  b'32313630363130323434303030303341202020204750535232504332303059573238303231313039'
CHKSUM:  b'EDBC'
Calc CHKSUM:  EDBC
BMS Serial Number: 216061024400003
Pack Serial Number: GPSR2PC200YW28
SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'10F0'
 - LCHKSUM:  49
 - LENID:  b'0F0'
 - LENID (int):  240
INFO:  b'0002100D020D030D020D050D030D030D040D020D050D030D030D040D030D060D030D01060C150C0A0C120C080C3E0C5FFEB3D0344904036D60008D6D60100CFF0CFF0CFE0D000D040D010D000CFF0D090D000D060D030D040D010D010CFF060C090C000C0B
CHKSUM:  b'C9D0'
Calc CHKSUM:  C9D0
Packs: 2
Pack 1, Total cells: 16
Pack 1, V Cell1: 3330 mV
Pack 1, V Cell2: 3331 mV
Pack 1, V Cell3: 3330 mV
Pack 1, V Cell4: 3333 mV
Pack 1, V Cell5: 3331 mV
Pack 1, V Cell6: 3331 mV
Pack 1, V Cell7: 3332 mV
Pack 1, V Cell8: 3330 mV
Pack 1, V Cell9: 3333 mV
Pack 1, V Cell10: 3331 mV
Pack 1, V Cell11: 3331 mV
Pack 1, V Cell12: 3332 mV
Pack 1, V Cell13: 3331 mV
Pack 1, V Cell14: 3334 mV
Pack 1, V Cell15: 3331 mV
Pack 1, V Cell16: 3329 mV
Pack 1, Cell Max Diff Volt Calc: 5 mV
Pack 1, Total temperature sensors: 6
Pack 1, Temp1: 36.3 ℃
Pack 1, Temp2: 35.2 ℃
Pack 1, Temp3: 36.0 ℃
Pack 1, Temp4: 35.0 ℃
Pack 1, Temp5: 40.4 ℃
Pack 1, Temp6: 43.7 ℃
Pack 1, I Pack: -3.32 A
Pack 1, V Pack: 53.3 V
Pack 1, I Remaining Capacity: 186920 mAh
Pack 1, I Full Capacity: 280000 mAh
Pack 1, SOC: 66.76 %
Pack 1, Cycles: 141
Pack 1, Design Capacity: 280000 mAh
Pack 1, SOH: 100.0 %
Error parsing BMS analog data: Cannot read multiple packs
Error retrieving BMS analog data: Error parsing BMS analog data: Cannot read multiple packs
SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'400C'
 - LCHKSUM:  52
 - LENID:  b'00C'
 - LENID (int):  12
INFO:  b'49046D606D60'
CHKSUM:  b'FB06'
Calc CHKSUM:  FB06
Pack Remaining Capacity: 186920 mAh
Pack Full Capacity: 280000 mAh
Pack Design Capacity: 280000 mAh
Pack SOC: 66.76 %
Pack SOH: 100.0 %
Script exiting
```

Tertiush, 2023-07-30:
Are you running the main or dev version? If the dev, then I just pushed an update that might fix this issue. Please run that one and check

tuxedo0801, 2023-08-03:
Had the DEV... will update and check again.

Tertiush, 2023-08-04:
Out of curiosity, did you move to the Dev addon yourself or did it somehow happen automatically? Asking because I added the dev version last week due to another issue, just afraid existing installations are somehow changing to that version....

closed 2023-08-11 by Tertiush