Tertiush/bmspace #23: Enersol Pace based Battery not reading correctly on 2 pack

countingteeth, 2023-09-03:
Hi,

I have a pack of 2 enersol batteries which use the Pace BMS firmware.  The second pack is not reading values correctly.  These values are correct in the Pace BMS software and I have "swapped" the rs232 around between the batteries and it always seems the issue is on the 2nd pack ie the slave.

Debug outpur

INFO:  b'0002100DAA0DA30DA70DAB0DA20D9A0DA40DA20DA40DA40DA50DAD0DA20DA70DA80DA7060B760B7A0B7F0B7D0B8E0B810000DA5326FF0227100008271064271027102710271064100DA60DAA0DAA0DAB0DA80DA70DAA0DA70DA80D9E0DA90DAA0DA60DA60DA60DA3060B5F0B6A0B690B6A0B7B0B6E0000DA7326FF0227100007271064271027102710271000'
CHKSUM:  b'BFFE'
Calc CHKSUM:  BFFE
Packs: 2
Pack 1, Total cells: 16
Pack 1, V Cell1: 3498 mV
Pack 1, V Cell2: 3491 mV
Pack 1, V Cell3: 3495 mV
Pack 1, V Cell4: 3499 mV
Pack 1, V Cell5: 3490 mV
Pack 1, V Cell6: 3482 mV
Pack 1, V Cell7: 3492 mV
Pack 1, V Cell8: 3490 mV
Pack 1, V Cell9: 3492 mV
Pack 1, V Cell10: 3492 mV
Pack 1, V Cell11: 3493 mV
Pack 1, V Cell12: 3501 mV
Pack 1, V Cell13: 3490 mV
Pack 1, V Cell14: 3495 mV
Pack 1, V Cell15: 3496 mV
Pack 1, V Cell16: 3495 mV
Pack 1, Cell Max Diff Volt Calc: 19 mV
Pack 1, Total temperature sensors: 6
Pack 1, Temp1: 20.4 ℃
Pack 1, Temp2: 20.8 ℃
Pack 1, Temp3: 21.3 ℃
Pack 1, Temp4: 21.1 ℃
Pack 1, Temp5: 22.8 ℃
Pack 1, Temp6: 21.5 ℃
Pack 1, I Pack: 0.0 A
Pack 1, V Pack: 55.891 V
Pack 1, I Remaining Capacity: 99830 mAh
Pack 1, I Full Capacity: 100000 mAh
Pack 1, SOC: 99.83 %
Pack 1, Cycles: 8
Pack 1, Design Capacity: 100000 mAh
Pack 1, SOH: 100.0 %
Pack 2, Total cells: 16
Pack 2, V Cell1: 10000 mV
Pack 2, V Cell2: 10000 mV
Pack 2, V Cell3: 10000 mV
Pack 2, V Cell4: 25616 mV
Pack 2, V Cell5: 3494 mV
Pack 2, V Cell6: 3498 mV
Pack 2, V Cell7: 3498 mV
Pack 2, V Cell8: 3499 mV
Pack 2, V Cell9: 3496 mV
Pack 2, V Cell10: 3495 mV
Pack 2, V Cell11: 3498 mV
Pack 2, V Cell12: 3495 mV
Pack 2, V Cell13: 3496 mV
Pack 2, V Cell14: 3486 mV
Pack 2, V Cell15: 3497 mV
Pack 2, V Cell16: 3498 mV
Pack 2, Cell Max Diff Volt Calc: 22130 mV
Pack 2, Total temperature sensors: 13
Pack 2, Temp1: 3977.9 ℃
Pack 2, Temp2: 3977.9 ℃
Pack 2, Temp3: 3977.9 ℃
Pack 2, Temp4: 3900.4 ℃
Pack 2, Temp5: 18.1 ℃
Pack 2, Temp6: 19.2 ℃
Pack 2, Temp7: 19.1 ℃
Pack 2, Temp8: 19.2 ℃
Pack 2, Temp9: 20.9 ℃
Pack 2, Temp10: 19.6 ℃
Pack 2, Temp11: -273.0 ℃
Pack 2, Temp12: 5319.3 ℃
Pack 2, Temp13: 725.3 ℃
Pack 2, I Pack: 5.51 A
Pack 2, V Pack: 4.096 V
Pack 2, I Remaining Capacity: 18310 mAh
Pack 2, I Full Capacity: 256390 mAh
Pack 2, SOC: 7.14 %
Pack 2, Cycles: 4135
Pack 2, Design Capacity: 41350 mAh
Pack 2, SOH: 620.05 %
Script running....
-> Outgoing Data:  b'~250146A60000FD97\r'
<- Incoming data:  b'~25014600400C26FF27102710FB0F\r'
SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'400C'
 - LCHKSUM:  52
 - LENID:  b'00C'
 - LENID (int):  12
INFO:  b'26FF27102710'
CHKSUM:  b'FB0F'
Calc CHKSUM:  FB0F
Pack Remaining Capacity: 99830 mAh
Pack Full Capacity: 100000 mAh
Pack Design Capacity: 100000 mAh
Pack SOC: 99.83 %
Pack SOH: 100.0 %

Tertiush, 2023-09-03:
It seems to be an issue with the bytes received between successive packs'
info. Please set the debug log to 3 and resend the logs. I need to see the
full byte stream received to find the exact issue

countingteeth, 2023-09-03:
Hi,

Thanks for the quick reply...

I've set the debug to 3....(This is from the log window in the plugin)

RTN:  b'00'
LENGTH:  b'B050'
 - LCHKSUM:  66
 - LENID:  b'050'
 - LENID (int):  80
INFO:  b'313739343032363331303030343436442020202034383130304D4231363233303332343033333620'
CHKSUM:  b'EDAA'
Calc CHKSUM:  EDAA
BMS Serial Number: 179402631000446
Pack Serial Number: 48100MB1623032
-> Outgoing Data:  b'~25014642E002FFFD05\r'
<- Incoming data:  b'~2501460061180002100DAA0DA40DA70DAB0DA30D9A0DA40DA20DA50DA40DA60DAE0DA20DA70DA80DA8060B720B740B790B780B8B0B7F0000DA5926F90227100008271064271027102710271064100DA60DAA0DAA0DAB0DA80DA70DAA0DA80DA80D9E0DA90DAA0DA60DA60DA70DA3060B5C0B670B660B670B7A0B6D0000DA7526F90227100007271064271027102710271000C03D\r'
SOI:  0x7e
VER:  b'25'
ADR:  b'01'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'6118'
 - LCHKSUM:  54
 - LENID:  b'118'
 - LENID (int):  280
INFO:  b'0002100DAA0DA40DA70DAB0DA30D9A0DA40DA20DA50DA40DA60DAE0DA20DA70DA80DA8060B720B740B790B780B8B0B7F0000DA5926F90227100008271064271027102710271064100DA60DAA0DAA0DAB0DA80DA70DAA0DA80DA80D9E0DA90DAA0DA60DA60DA70DA3060B5C0B670B660B670B7A0B6D0000DA7526F90227100007271064271027102710271000'
CHKSUM:  b'C03D'
Calc CHKSUM:  C03D
Packs: 2
Pack 1, Total cells: 16
Pack 1, V Cell1: 3498 mV
Pack 1, V Cell2: 3492 mV
Pack 1, V Cell3: 3495 mV
Pack 1, V Cell4: 3499 mV
Pack 1, V Cell5: 3491 mV
Pack 1, V Cell6: 3482 mV
Pack 1, V Cell7: 3492 mV
Pack 1, V Cell8: 3490 mV
Pack 1, V Cell9: 3493 mV
Pack 1, V Cell10: 3492 mV
Pack 1, V Cell11: 3494 mV
Pack 1, V Cell12: 3502 mV
Pack 1, V Cell13: 3490 mV
Pack 1, V Cell14: 3495 mV
Pack 1, V Cell15: 3496 mV
Pack 1, V Cell16: 3496 mV
Pack 1, Cell Max Diff Volt Calc: 20 mV
Pack 1, Total temperature sensors: 6
Pack 1, Temp1: 20.0 ℃
Pack 1, Temp2: 20.2 ℃
Pack 1, Temp3: 20.7 ℃
Pack 1, Temp4: 20.6 ℃
Pack 1, Temp5: 22.5 ℃
Pack 1, Temp6: 21.3 ℃
Pack 1, I Pack: 0.0 A
Pack 1, V Pack: 55.897 V
Pack 1, I Remaining Capacity: 99770 mAh
Pack 1, I Full Capacity: 100000 mAh
Pack 1, SOC: 99.77 %
Pack 1, Cycles: 8
Pack 1, Design Capacity: 100000 mAh
Pack 1, SOH: 100.0 %
Pack 2, Total cells: 16
Pack 2, V Cell1: 10000 mV
Pack 2, V Cell2: 10000 mV
Pack 2, V Cell3: 10000 mV
Pack 2, V Cell4: 25616 mV
Pack 2, V Cell5: 3494 mV
Pack 2, V Cell6: 3498 mV
Pack 2, V Cell7: 3498 mV
Pack 2, V Cell8: 3499 mV
Pack 2, V Cell9: 3496 mV
Pack 2, V Cell10: 3495 mV
Pack 2, V Cell11: 3498 mV
Pack 2, V Cell12: 3496 mV
Pack 2, V Cell13: 3496 mV
Pack 2, V Cell14: 3486 mV
Pack 2, V Cell15: 3497 mV
Pack 2, V Cell16: 3498 mV
Pack 2, Cell Max Diff Volt Calc: 22130 mV
Pack 2, Total temperature sensors: 13
Pack 2, Temp1: 3977.9 ℃
Pack 2, Temp2: 3977.9 ℃
Pack 2, Temp3: 4003.5 ℃
Pack 2, Temp4: 3900.4 ℃
Pack 2, Temp5: 17.8 ℃
Pack 2, Temp6: 18.9 ℃
Pack 2, Temp7: 18.8 ℃
Pack 2, Temp8: 18.9 ℃
Pack 2, Temp9: 20.8 ℃
Pack 2, Temp10: 19.5 ℃
Pack 2, Temp11: -273.0 ℃
Pack 2, Temp12: 5319.5 ℃
Pack 2, Temp13: 724.7 ℃
Pack 2, I Pack: 5.51 A
Pack 2, V Pack: 4.096 V
Pack 2, I Remaining Capacity: 18310 mAh
Pack 2, I Full Capacity: 256390 mAh
Pack 2, SOC: 7.14 %
Pack 2, Cycles: 4135
Pack 2, Design Capacity: 41350 mAh
Pack 2, SOH: 620.05 %
Script running....

Tertiush, 2023-09-03:
Thanks. Are you running the main or dev version? I'll try to have a look
later, if time permits

countingteeth, 2023-09-03:
Hi,

Current version: 2.2.0 

tks

Tertiush, 2023-09-03:
Your BMS' data has some unexpected bytes between the packs' data. Please install the dev version and set the force_pack_offset option to 20. This is the 20 bytes that for some reason is in-between the pack's data. Test and let me know pls.

countingteeth, 2023-09-03:
Hi,

Loos like it it doing its thing now...

Hello BMS Pace
Starting up...
Loading options.json
Config: {"mqtt_host": "192.168.9.99", "mqtt_port": 1883, "mqtt_user": "homeassistant", "mqtt_password": "eXei7ohfooguawee0fahgh5echieseiZ8aiYahse3taivoh4biv5paigequee8ja", "mqtt_ha_discovery": true, "mqtt_ha_discovery_topic": "homeassistant", "mqtt_base_topic": "bmspace", "connection_type": "Serial", "bms_ip": "10.0.0.161", "bms_port": 5000, "bms_serial": "/dev/ttyUSB0", "scan_interval": 5, "debug_output": 0, "force_pack_offset": 20}
Connection Type: Serial
MQTT connected with result code 0
Connecting to BMS...
trying to connect /dev/ttyUSB0
BMS serial connected
BMS Version: P16S100A-17940-2.08T
BMS Serial Number: 179402631000446
Pack Serial Number: 48100MB1623032
Packs: 2
Pack 1, Total cells: 16
Pack 1, V Cell1: 3453 mV
Pack 1, V Cell2: 3449 mV
Pack 1, V Cell3: 3452 mV
Pack 1, V Cell4: 3454 mV
Pack 1, V Cell5: 3449 mV
Pack 1, V Cell6: 3443 mV
Pack 1, V Cell7: 3450 mV
Pack 1, V Cell8: 3449 mV
Pack 1, V Cell9: 3450 mV
Pack 1, V Cell10: 3450 mV
Pack 1, V Cell11: 3451 mV
Pack 1, V Cell12: 3456 mV
Pack 1, V Cell13: 3448 mV
Pack 1, V Cell14: 3452 mV
Pack 1, V Cell15: 3452 mV
Pack 1, V Cell16: 3451 mV
Pack 1, Cell Max Diff Volt Calc: 13 mV
Pack 1, Total temperature sensors: 6
Pack 1, Temp1: 22.5 ℃
Pack 1, Temp2: 23.1 ℃
Pack 1, Temp3: 23.5 ℃
Pack 1, Temp4: 23.4 ℃
Pack 1, Temp5: 24.8 ℃
Pack 1, Temp6: 23.4 ℃
Pack 1, I Pack: 0.0 A
Pack 1, V Pack: 55.209 V
Pack 1, I Remaining Capacity: 99810 mAh
Pack 1, I Full Capacity: 100000 mAh
Pack 1, SOC: 99.81 %
Pack 1, Cycles: 8
Pack 1, Design Capacity: 100000 mAh
Pack 1, SOH: 100.0 %
Pack 2, Total cells: 16
Pack 2, V Cell1: 3449 mV
Pack 2, V Cell2: 3453 mV
Pack 2, V Cell3: 3455 mV
Pack 2, V Cell4: 3454 mV
Pack 2, V Cell5: 3453 mV
Pack 2, V Cell6: 3453 mV
Pack 2, V Cell7: 3454 mV
Pack 2, V Cell8: 3452 mV
Pack 2, V Cell9: 3452 mV
Pack 2, V Cell10: 3450 mV
Pack 2, V Cell11: 3453 mV
Pack 2, V Cell12: 3454 mV
Pack 2, V Cell13: 3452 mV
Pack 2, V Cell14: 3451 mV
Pack 2, V Cell15: 3451 mV
Pack 2, V Cell16: 3447 mV
Pack 2, Cell Max Diff Volt Calc: 8 mV
Pack 2, Total temperature sensors: 6
Pack 2, Temp1: 20.1 ℃
Pack 2, Temp2: 21.5 ℃
Pack 2, Temp3: 21.3 ℃
Pack 2, Temp4: 21.4 ℃
Pack 2, Temp5: 22.8 ℃
Pack 2, Temp6: 21.5 ℃
Pack 2, I Pack: 0.0 A
Pack 2, V Pack: 55.233 V
Pack 2, I Remaining Capacity: 100000 mAh
Pack 2, I Full Capacity: 100000 mAh
Pack 2, SOC: 100.0 %
Pack 2, Cycles: 7
Pack 2, Design Capacity: 100000 mAh
Pack 2, SOH: 100.0 %
Script running....
Pack Remaining Capacity: 99810 mAh
Pack Full Capacity: 100000 mAh
Pack Design Capacity: 100000 mAh
Pack SOC: 99.81 %
Pack SOH: 100.0 %
Packs for warnings: 2
Pack 1, warnings: 
Pack 1, balancing1: 00000000
Pack 1, balancing2: 00000000
Pack 2, warnings: Protection State 2: Fully
Pack 2, balancing1: 00000000
Pack 2, balancing2: 00000000

Tertiush, 2023-09-03:
Great, just edit your logs above and remove any sensitive info......

countingteeth, 2023-09-03:
tks again .....

closed 2023-09-03 by countingteeth