Tertiush/bmspace #25: No config file found (docker container)

jogybaer0815, 2023-12-07:
hello and thank you for your work. I have successfully setup bmspace under HomeAssistant but i prefer to build a Docker container. 
I have edited the config.yaml but the created container do not find any config file: 'No config file found'
What can i do? Thanks in advance

jogybaer0815, 2023-12-07:
i've got it.
Dockerfile have to expand with 'COPY config.yaml .'

jpmeijers, 2023-12-08:
I also prefer running directly in Docker rather than through HA. I use the Docker Compose file, which mounts in the config file as a volume. 

See:
https://github.com/Tertiush/bmspace/blob/b4452ad7c88f040f6a85ce694163af6a836d14d7/docker-compose.yaml#L9

jogybaer0815, 2023-12-08:
Thanks for your reply. I'm able to build the container but there are now some errors:

Pack 1, I Remaining Capacity: 101160 mAh
Pack 1, I Full Capacity: 101390 mAh
Pack 1, SOC: 99.77 %
Pack 1, Cycles: 3
Pack 1, Design Capacity: 100000 mAh
Pack 1, SOH: 101.39 %
Pack 2, Total cells: 15
Pack 2, V Cell1: 3342 mV
Pack 2, V Cell2: 3342 mV
Pack 2, V Cell3: 3343 mV
Pack 2, V Cell4: 3342 mV
Pack 2, V Cell5: 3342 mV
Pack 2, V Cell6: 3342 mV
Pack 2, V Cell7: 3343 mV
Pack 2, V Cell8: 3342 mV
Pack 2, V Cell9: 3343 mV
Pack 2, V Cell10: 3342 mV
Pack 2, V Cell11: 3343 mV
Pack 2, V Cell12: 3343 mV
Pack 2, V Cell13: 3343 mV
Pack 2, V Cell14: 3341 mV
Pack 2, V Cell15: 3342 mV
Pack 2, Cell Max Diff Volt Calc: 2 mV
Pack 2, Total temperature sensors: 6
Pack 2, Temp1: 16.9 ℃
Pack 2, Temp2: 17.1 ℃
Pack 2, Temp3: 17.4 ℃
Pack 2, Temp4: 17.0 ℃
Pack 2, Temp5: 19.1 ℃
Pack 2, Temp6: 18.3 ℃
Pack 2, I Pack: 0.0 A
Pack 2, V Pack: 50.104 V
Pack 2, I Remaining Capacity: 96120 mAh
Pack 2, I Full Capacity: 100500 mAh
Pack 2, SOC: 95.64 %
Pack 2, Cycles: 4
Pack 2, Design Capacity: 100000 mAh
Pack 2, SOH: 100.5 %
Pack 3, Total cells: 15
Pack 3, V Cell1: 3340 mV
Pack 3, V Cell2: 3339 mV
Pack 3, V Cell3: 3340 mV
Pack 3, V Cell4: 3340 mV
Pack 3, V Cell5: 3341 mV
Pack 3, V Cell6: 3340 mV
Pack 3, V Cell7: 3337 mV
Pack 3, V Cell8: 3341 mV
Pack 3, V Cell9: 3340 mV
Pack 3, V Cell10: 3340 mV
Pack 3, V Cell11: 3341 mV
Pack 3, V Cell12: 3340 mV
Pack 3, V Cell13: 3341 mV
Pack 3, V Cell14: 3341 mV
Pack 3, V Cell15: 3341 mV
Pack 3, Cell Max Diff Volt Calc: 4 mV
Pack 3, Total temperature sensors: 6
Pack 3, Temp1: 16.6 ℃
Pack 3, Temp2: 17.0 ℃
Pack 3, Temp3: 16.8 ℃
Pack 3, Temp4: 17.1 ℃
Pack 3, Temp5: 19.0 ℃
Pack 3, Temp6: 18.8 ℃
Pack 3, I Pack: 0.0 A
Pack 3, V Pack: 50.096 V
Pack 3, I Remaining Capacity: 96160 mAh
Pack 3, I Full Capacity: 100500 mAh
Pack 3, SOC: 95.68 %
Pack 3, Cycles: 4
Pack 3, Design Capacity: 100000 mAh
Pack 3, SOH: 100.5 %
Script running....
Pack Remaining Capacity: 30 mAh
Pack Full Capacity: 38530 mAh
Pack Design Capacity: 261250 mAh
Pack SOC: 0.08 %
Pack SOH: 14.75 %
Packs for warnings: 3
Error parsing BMS warning data:  b'9B'
Error retrieving BMS warning info: Error parsing BMS warning data: b'9B'
Publishing HA Discovery topic...
Error parsing BMS analog data:  invalid literal for int() with base 16: b''
Error retrieving BMS analog data: Error parsing BMS analog data: invalid literal for int() with base 16: b''
Error parsing BMS warning data:  b'9B'
Error retrieving BMS warning info: Error parsing BMS warning data: b'9B'
...

closed 2024-01-28 by github-actions[bot]