Tertiush/bmspace #26: Error parsing BMS analog data: Cannot read multiple packs

asdt1803, 2023-12-13:
Hello,

Thnk you for this greatsoftware. It works out of the box with my Gobel Power GP-SR1-PC200 packs.
Only thing that does not work is reading multiple packs. If connected an USB-Serial adapter to the master pack and can read information for the master battery but not for the 3 slave packs:

trying to connect /dev/ttyUSB0
BMS serial connected
BMS Version: P16S200A-21606-1.01
BMS Serial Number: xxxx
Pack Serial Number: xxxx
Packs: 4
Pack 1, Total cells: 16
Pack 1, V Cell1: 3328 mV
Pack 1, V Cell2: 3328 mV
Pack 1, V Cell3: 3327 mV
Pack 1, V Cell4: 3328 mV
Pack 1, V Cell5: 3328 mV
Pack 1, V Cell6: 3328 mV
Pack 1, V Cell7: 3328 mV
Pack 1, V Cell8: 3328 mV
Pack 1, V Cell9: 3328 mV
Pack 1, V Cell10: 3328 mV
Pack 1, V Cell11: 3328 mV
Pack 1, V Cell12: 3328 mV
Pack 1, V Cell13: 3328 mV
Pack 1, V Cell14: 3328 mV
Pack 1, V Cell15: 3330 mV
Pack 1, V Cell16: 3328 mV
Pack 1, Cell Max Diff Volt Calc: 3 mV
Pack 1, Total temperature sensors: 6
Pack 1, Temp1: 15.0 ℃
Pack 1, Temp2: 14.8 ℃
Pack 1, Temp3: 14.6 ℃
Pack 1, Temp4: 15.1 ℃
Pack 1, Temp5: 16.4 ℃
Pack 1, Temp6: 23.6 ℃
Pack 1, I Pack: 0.0 A
Pack 1, V Pack: 53.249 V
Pack 1, I Remaining Capacity: 308000 mAh
Pack 1, I Full Capacity: 308000 mAh
Pack 1, SOC: 100.0 %
Pack 1, Cycles: 5
Pack 1, Design Capacity: 280000 mAh
Pack 1, SOH: 110.0 %
Error parsing BMS analog data: Cannot read multiple packs
Error retrieving BMS analog data: Error parsing BMS analog data: Cannot read multiple packs

Am I doing something wrong, is this possiblya limitation of the BMS or a problem in bms.py?

Thank you for your help
Andre

asdt1803, 2023-12-13:
Found the dev version is working fine with multiple packs.

closed 2023-12-13 by asdt1803

Tertiush, 2023-12-14:
Awesome, was about to suggest that, and playing with the "force offset"
configuration value.

Glad its working