Tertiush/bmspace #27: division by zero in BMS analog data

GerhardLang, 2024-01-17:
Hello.
I do have 10 Gobel Batteries, ID 2 is actually not connected.

The 2.2.3 dev Version is installed, 2.2.2 did not work even with the first Battery ID.

It is recognized:
BMS serial connected
BMS Version: P16S200A-21606-2.02 

Then, the Pack 1 is read. Pack 2 is not existent, so 0 ist o.k.
But after this and the division by 0 no further packs are read.

Do you have a clue what I can do to get all existing 9 into HA?

Thanks.

....
Pack 1, V Cell3: 3285 mV
Pack 1, V Cell4: 3286 mV
Pack 1, V Cell5: 3288 mV
Pack 1, V Cell6: 3287 mV
Pack 1, V Cell7: 3286 mV
Pack 1, V Cell8: 3287 mV
Pack 1, V Cell9: 3286 mV
Pack 1, V Cell10: 3286 mV
Pack 1, V Cell11: 3287 mV
Pack 1, V Cell12: 3283 mV
Pack 1, V Cell13: 3285 mV
Pack 1, V Cell14: 3286 mV
Pack 1, V Cell15: 3287 mV
Pack 1, V Cell16: 3287 mV
Pack 1, Cell Max Diff Volt Calc: 5 mV
Pack 1, Total temperature sensors: 6
Pack 1, Temp1: 9.9 ℃
Pack 1, Temp2: 9.9 ℃
Pack 1, Temp3: 9.7 ℃
Pack 1, Temp4: 9.8 ℃
Pack 1, Temp5: 11.0 ℃
Pack 1, Temp6: 12.1 ℃
Pack 1, I Pack: -4.08 A
Pack 1, V Pack: 52.578 V
Pack 1, I Remaining Capacity: 174680 mAh
Pack 1, I Full Capacity: 279460 mAh
Pack 1, SOC: 62.51 %
Pack 1, Cycles: 19
Pack 1, Design Capacity: 280000 mAh
Pack 1, SOH: 99.81 %
Pack 2, Total cells: 16
Pack 2, V Cell1: 0 mV
Pack 2, V Cell2: 0 mV
Pack 2, V Cell3: 0 mV
Pack 2, V Cell4: 0 mV
Pack 2, V Cell5: 0 mV
Pack 2, V Cell6: 0 mV
Pack 2, V Cell7: 0 mV
Pack 2, V Cell8: 0 mV
Pack 2, V Cell9: 0 mV
Pack 2, V Cell10: 0 mV
Pack 2, V Cell11: 0 mV
Pack 2, V Cell12: 0 mV
Pack 2, V Cell13: 0 mV
Pack 2, V Cell14: 0 mV
Pack 2, V Cell15: 0 mV
Pack 2, V Cell16: 0 mV
Pack 2, Cell Max Diff Volt Calc: 0 mV
Pack 2, Total temperature sensors: 6
Pack 2, Temp1: 0.0 ℃
Pack 2, Temp2: 0.0 ℃
Pack 2, Temp3: 0.0 ℃
Pack 2, Temp4: 0.0 ℃
Pack 2, Temp5: 0.0 ℃
Pack 2, Temp6: 0.0 ℃
Pack 2, I Pack: 0.0 A
Pack 2, V Pack: 0.0 V
Pack 2, I Remaining Capacity: 0 mAh
Pack 2, I Full Capacity: 0 mAh
Error parsing BMS analog data:  division by zero
Error retrieving BMS analog data: Error parsing BMS analog data: division by zero
Pack Remaining Capacity: 174670 mAh
Pack Full Capacity: 279460 mAh
Pack Design Capacity: 280000 mAh
Pack SOC: 62.5 %
Pack SOH: 99.81 %
Packs for warnings: 10
Pack 1, warnings: 
Pack 1, balancing1: 00000000
Pack 1, balancing2: 00000000
Pack 2, warnings: cell 1 < low limit, cell 2 < low limit, cell 3 < low limit, cell 4 < low limit, cell 5 < low limit, cell 6 < low limit, cell 7 < low limit, cell 8 < low limit, cell 9 < low limit, cell 10 < low limit, cell 11 < low limit, cell 12 < low limit, cell 13 < low limit, cell 14 < low limit, cell 15 < low limit, cell 16 < low limit, total voltage < low limit
Pack 2, balancing1: 00000000
Pack 2, balancing2: 00000000
Pack 3, warnings: 
Pack 3, balancing1: 00000000
Pack 3, balancing2: 00000000
Pack 4, warnings: 
Pack 4, balancing1: 00000000
Pack 4, balancing2: 00000000
Pack 5, warnings: 
Pack 5, balancing1: 00000000
Pack 5, balancing2: 00000000
Pack 6, warnings: 
Pack 6, balancing1: 00000000
Pack 6, balancing2: 00000000
Pack 7, warnings: 
Pack 7, balancing1: 00000000
Pack 7, balancing2: 00000000
Pack 8, warnings: 
Pack 8, balancing1: 00000000
Pack 8, balancing2: 00000000
Pack 9, warnings: 
Pack 9, balancing1: 00000000
Pack 9, balancing2: 00000000
Pack 10, warnings: Warning State 2: Low power warn
Pack 10, balancing1: 00000000
Pack 10, balancing2: 00000000
Publishing HA Discovery topic...
Error parsing BMS analog data:  division by zero
Error retrieving BMS analog data: Error parsing BMS analog data: division by zero

Tertiush, 2024-01-27:
If its not to much hassle, can you restart the addon with maximum debug levels and one of your packs turned off, to cause the error? I want to see at which place in the code it occurs (hoping the more verbose log will show where).

GerhardLang, 2024-01-29:
Hello again.
I did switch the 2nd pack off like when the original error occured.
Here is the file with the division by zero.
I hope this helps.

[1f71505a_bms_pace_dev_2024-01-29T17-27-07.646Z.log](https://github.com/Tertiush/bmspace/files/14088211/1f71505a_bms_pace_dev_2024-01-29T17-27-07.646Z.log)

GerhardLang, 2024-02-11:
@Tertiush 
Hi, did you have time to look if my log does show any hint for the error?
How can I support you, I did not find a link for that.

Tertiush, 2024-02-14:
Hi, no haven't had time yet. Will likely only again do an update in about 3
weeks' time due to other commitments. Generally the code is stable for all
others so I look at issues when time permits. ;)

GerhardLang, 2024-02-15:
Thank you, all fine.

GerhardLang, 2024-02-25:
Thank you for your effort. Will close the issue and open a new issue (hopefully not) if the error will occur again.
@Tertiush did you include the hint of @asdt1803 when correcting the code?
Have a good time.

GerhardLang, 2024-02-25:
Error actually not ocurred anymore.

closed 2024-02-25 by GerhardLang