Tertiush/bmspace #28: Every few days script stops with errors

GerhardLang, 2024-01-23:
Hi, 

Today with fully charged battery I got the following script error.
@Tertiush maybe you can check what this could be?

Pack 10, balancing1: 00000000
Pack 10, balancing2: 00000000
Publishing HA Discovery topic...
Error parsing BMS analog data:  invalid literal for int() with base 16: b''
Error retrieving BMS analog data: Error parsing BMS analog data: invalid literal for int() with base 16: b''
Error parsing BMS analog data:  invalid literal for int() with base 16: b''
Error retrieving BMS analog data: Error parsing BMS analog data: invalid literal for int() with base 16: b''
Error parsing BMS analog data:  invalid literal for int() with base 16: b''
Error retrieving BMS analog data: Error parsing BMS analog data: invalid literal for int() with base 16: b''

and so on. After restarting it works again (every few days this problem occurs).

After restart:

Pack Remaining Capacity: 279430 mAh
Pack Full Capacity: 279430 mAh
Pack Design Capacity: 280000 mAh
Pack SOC: 100.0 %
Pack SOH: 99.8 %
Packs for warnings: 10
Pack 1, warnings: Protection State 2: Fully
Pack 1, balancing1: 00000000
Pack 1, balancing2: 00000000
Pack 2, warnings: Protection State 2: Fully
Pack 2, balancing1: 11111011
Pack 2, balancing2: 11111110
Pack 3, warnings: Protection State 2: Fully
Pack 3, balancing1: 00000000
Pack 3, balancing2: 00000000
Pack 4, warnings: Protection State 2: Fully
Pack 4, balancing1: 00000000
Pack 4, balancing2: 00000000
Pack 5, warnings: Protection State 2: Fully
Pack 5, balancing1: 00000000
Pack 5, balancing2: 00000000
Pack 6, warnings: Protection State 2: Fully
Pack 6, balancing1: 00000000
Pack 6, balancing2: 00000000
Pack 7, warnings: Protection State 2: Fully
Pack 7, balancing1: 00000000
Pack 7, balancing2: 00000000
Pack 8, warnings: Protection State 2: Fully
Pack 8, balancing1: 00000000
Pack 8, balancing2: 00000000
Pack 9, warnings: Protection State 2: Fully
Pack 9, balancing1: 00000000
Pack 9, balancing2: 00000000
Pack 10, warnings: Protection State 2: Fully
Pack 10, balancing1: 00000000
Pack 10, balancing2: 00000000
Publishing HA Discovery topic...

Thank you.

Tertiush, 2024-01-23:
Hi hi, which version are you using?

Then, please use the maximum debug level to see which part of the code is throwing this error. I can then build better error handling into that portion.

GerhardLang, 2024-01-24:
Will do and post as soon as the error is back.

GerhardLang, 2024-02-05:
The error did not occur anymore. So this is not reproduceable. So only div by 0 is still open see other issue.

closed 2024-02-05 by GerhardLang

asdt1803, 2024-02-19:
Hello @Tertiush ,

I experience the same problem with my 4 Gobel Batteries as @GerhardLang does. This happens sporadically a few times a day.  I added some additional debug output to the script and it seems that sometimes the script (2.2.3dev) starts reading the cell voltages of my third pack at the wrong position in inc_data (6 bytes early). Please see attached debug log. Subsequently the figures for pack 4 are also implausible. The incoming data seems to be valid, though.

I was not yet able to identify the root cause of the problem but I found a quick and dirty hack that solves the problem for me by correcting the byte_index before the script starts iterating the cells voltages:

            byte_corrected = 6 + ((p-1) * 146)
            if byte_index != byte_corrected:
                print("Wrong start byte index Pack: " +str(p) + " => " + str(byte_index) + " instead " + str(byte_corrected))
                byte_index = byte_corrected

Output in case the problem occurs: Wrong start byte index Pack: 3 => 292 instead 298

Thank you for your help

[pace-bms-dev.log](https://github.com/Tertiush/bmspace/files/14327862/pace-bms-dev.log)

Tertiush, 2024-03-02:
I think different vendors/versions of packs have slightly different firmwares which sporadically causes these issues. I had them before and fixed it using a while loop to find the cell count number preceding the cells values. When I find this value, e.g. 16 the parsing starts. Its likely that this value is sometimes present for another reason causing this false offset. Your fix will work for your firmware and cell count, but may not for all versions & cell counts out there.... TBH I don't have time to release special versions for different packs as its working fine for me and most out there. Can I suggest you make a fork, apply the fixes and release that to those with the same issue?

asdt1803, 2024-03-02:
Hello Tertiush,

You are absolutely right, the workaround mentioned above  is specific for the BMS and firmware I'm using. I wrote this post before I found out what caused the problem and posted it hoping that might help you to identify the problem. 
Now I think I identified the problem and suggested an generic fix in https://github.com/Tertiush/bmspace/issues/32 that I think would make the workaround above obsolete.