Tertiush/bmspace #29: Numberin of packs and cell with 2 digits would help sorting

GerhardLang, 2024-01-25:
Hello again, sorry to bother you again @Tertiush 

I like your project very much. When visualizing the data which is sent by MQTT the ordering is not given, because the pack and cell numbers are not numbered
01 02 03 ... 10 11 12...
but
1 2 3 ... 10 11 12...
If you could adapt this in a next release (I am using the actual dev release) this would be fantastic for correct order of the graphs and lists.
Today it will list
10 11 12.. 16 1 2 3 .. 9
Thank you very much.

Tertiush, 2024-01-27:
Hi, I hear you, also hate it when this happens ;)

This can definitely be an option in the next version. I wouldn't want to hard code it as cell can go up to 16 x 16 = 256 I think, needing 2 leading zeros below 100. So having 0 prefix padding options 0,1,2 should do the job.

In the next dev version, pending some other errors I want to look at......

Tertiush, 2024-02-24:
New dev version being pushed now with this.

GerhardLang, 2024-02-25:
It works like a charm. 

Thousand thanks.

GerhardLang, 2024-02-25:
Great Work.

closed 2024-02-25 by GerhardLang