Tertiush/bmspace #33: Hardware Info and more documentation

Norm71, 2024-03-05:
I do have a System from Tewaycell [Tewaycell 48V 200Ah 10Kwh All-in-one Mobile ESS Built-in Hybrid Invert](https://tewaycell.com/products/tewaycell-48v-200ah-10kwh-all-in-one-mobile-ess-built-in-hybrid-inverter?variant=41313098858553) and they provided a RS232 to USB Cable with it.

They also sent the PACEexTool but its completely in Chinese.

Now i found your Home Assistant integration. 😀

Can you explain a bit more how i do get the basic connection?
I think i need to install the USB driver for Linux first on the Homeassistant Raspberry. Is that possible?
Then i have questions for the requirements:
paho_mqtt==1.6.1
PyYAML==6.0.1
pyserial==3.5

I dont know how to get the reuirements.

I have Mosquito MQTT and use it already for my OpenDTU and ESP32

I posted in Power Forum but did not get any answer. So i try directly here.

Tertiush, 2024-03-06:
Do you have home assistant operating system, or just home assistant core
installed? The addon is meant for operating system.

If you run core then you need to install the dependancies in the
requirements file, change the config file to suite your needs, and finally
run the script manually. If you need to run the script as a system service,
Google will help you out there.

Norm71, 2024-03-06:
Its Home Assistant Operating System. So with HA OS i dont need the paho_mqtt, PyYAML and pyserial?
Do you have some website where your setup is documented or do you know some?

Tertiush, 2024-03-07:
Nope you don't need it, HAOS will handle all of that for you. Just Google
how to install addons in home assistant, pretty straightforward.

Norm71, 2024-03-07:
Add on is installed but I don't get any signal from the USB port. Don't know how to find out which is used in Home assistant

Norman Brodowsky
Herzogst.21
41747 Viersen
01522 3640871

Tertiush, 2024-03-07:
Go look in the system menu for hardware info or something like that. Should
be usbtty1 or something similar. Case sensitive

Norm71, 2024-03-07:
Thats it:
Hello BMS Pace
Starting up...
Loading options.json
Config: {"mqtt_host": "192.168.17.133", "mqtt_port": 1883, "mqtt_user": "mqtt-user", "mqtt_password": "jh99arfmn98fra", "mqtt_ha_discovery": true, "mqtt_ha_discovery_topic": "homeassistant", "mqtt_base_topic": "bmspace", "connection_type": "Serial", "bms_ip": "10.0.0.161", "bms_port": 5000, "bms_serial": "/dev/ttyACM1", "scan_interval": 5, "debug_output": 0}
Connection Type: Serial
MQTT connected with result code 0
Connecting to BMS...
trying to connect /dev/ttyACM1
BMS serial connected

Norm71, 2024-03-07:
Sorry can you remove my address out of my reply ? Answered from phone and did not thing about it

Norm71, 2024-03-07:
It was to early to be happy. That was my Z-Wave Stick. Cable is not listed will try a Waveshare RS232 RS485 to WiFi POE Ethernet Serial Server Converter. 
Is there any Hardware you can recommend?

Tertiush, 2024-03-09:
Nope sorry, just used what I has laying around.

closed 2024-03-17 by Tertiush