Tertiush/bmspace #34: USB to Serial connection error

mcleanleeroy, 2024-03-17:
Hi Tertiush

I am running home assistant OS installed in an Intel NUC with the latest versions
Core
2024.3.1
Supervisor
2024.03.0
Operating System
12.1
Frontend
20240307.0

I excitedly installed your Home assistant addon. I've checked my usb-serial adaptor is connected in the hardware to /dev/ttyUSB0. When I start the addon and look at the logs I get this message. Is something missing in the addon install? 

Hello BMS Pace
Starting up...
Loading options.json
Config: {"mqtt_host": "10.0.0.132", "mqtt_port": 1883, "mqtt_user": "mqtt-users", "mqtt_password": "mqtt-users", "mqtt_ha_discovery": true, "mqtt_ha_discovery_topic": "homeassistant", "mqtt_base_topic": "bmspace", "connection_type": "Serial", "bms_ip": "10.0.0.161", "bms_port": 5000, "bms_serial": "/dev/ttyUSB0", "scan_interval": 5, "debug_output": 3}
Connection Type: Serial
Traceback (most recent call last):
  File "/workdir/./bms.py", line 75, in <module>
    client.connect(config['mqtt_host'], config['mqtt_port'], 60)
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 914, in connect
    return self.reconnect()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1044, in reconnect
    sock = self._create_socket_connection()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
    return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
  File "/usr/local/lib/python3.9/socket.py", line 844, in create_connection
    raise err
  File "/usr/local/lib/python3.9/socket.py", line 832, in create_connection
    sock.connect(sa)
OSError: [Errno 113] Host is unreachable

Tertiush, 2024-03-17:
Is your MQTT host IP correct, still seeing the default 10.0.0.132?

mcleanleeroy, 2024-03-17:
I think I misunderstood your whole process. I thought the MQTT was only for connecting via the ethernet and wasn't used for serial. I have setup my MQTT service and all is working. Thank you for pointing me in the right direction.

closed 2024-03-17 by mcleanleeroy