Tertiush/bmspace #37: Not reading multiple packs , only the first one

xsherlockpl, 2024-04-05:
Hi, So I have a 3 pack from SUNPRO that I try to read with this library over RS-232. It only works for the first battery module out of 3.
The data is available as their PC software does read all 3 packs data from the single connection.
[image]
So I guess it must be something simple to fix.

Debug output is following. (A minior thing I notice here is that SN is truncated on the hypen)

`
MQTT connected with result code 0
Connecting to BMS...
trying to connect /dev/ttyUSB0
BMS serial connected
-> Outgoing Data:  b'~250146C10000FD9A\r'
<- Incoming data:  b'~25004600602850313653313030412D31313537302D322E303000F58B\r'
SOI:  0x7e
VER:  b'25'
ADR:  b'00'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'6028'
 - LCHKSUM:  54
 - LENID:  b'028'
 - LENID (int):  40
INFO:  b'50313653313030412D31313537302D322E303000'
CHKSUM:  b'F58B'
Calc CHKSUM:  F58B
BMS Version: P16S100A-11570-2.00
-> Outgoing Data:  b'~250146C20000FD99\r'
<- Incoming data:  b'~25004600B0503033202020202020202020202020202020202020393930323031363136313030332D323330363136EDE4\r'
SOI:  0x7e
VER:  b'25'
ADR:  b'00'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'B050'
 - LCHKSUM:  66
 - LENID:  b'050'
 - LENID (int):  80
INFO:  b'3033202020202020202020202020202020202020393930323031363136313030332D323330363136'
CHKSUM:  b'EDE4'
Calc CHKSUM:  EDE4
BMS Serial Number: 03
Pack Serial Number: 9902016161003-
-> Outgoing Data:  b'~25014642E002FFFD05\r'
<- Incoming data:  b'~25004600D07C0001100CCB0CCD0CCE0CCD0CCC0CCC0CCD0CCD0CCD0CCC0CCC0CCD0CCC0CCC0CCD0CCC060B6D0B5E0B600B540BC80BB90000CCFC10ED02417200033EE419E0F5\r'
SOI:  0x7e
VER:  b'25'
ADR:  b'00'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'D07C'
 - LCHKSUM:  68
 - LENID:  b'07C'
 - LENID (int):  124
INFO:  b'0001100CCB0CCD0CCE0CCD0CCC0CCC0CCD0CCD0CCD0CCC0CCC0CCD0CCC0CCC0CCD0CCC060B6D0B5E0B600B540BC80BB90000CCFC10ED02417200033EE419'
CHKSUM:  b'E0F5'
Calc CHKSUM:  E0F5
Packs: 1
Pack 1, Total cells: 16
Pack 1, V Cell1: 3275 mV
Pack 1, V Cell2: 3277 mV
Pack 1, V Cell3: 3278 mV
Pack 1, V Cell4: 3277 mV
Pack 1, V Cell5: 3276 mV
Pack 1, V Cell6: 3276 mV
Pack 1, V Cell7: 3277 mV
Pack 1, V Cell8: 3277 mV
Pack 1, V Cell9: 3277 mV
Pack 1, V Cell10: 3276 mV
Pack 1, V Cell11: 3276 mV
Pack 1, V Cell12: 3277 mV
Pack 1, V Cell13: 3276 mV
Pack 1, V Cell14: 3276 mV
Pack 1, V Cell15: 3277 mV
Pack 1, V Cell16: 3276 mV
Pack 1, Cell Max Diff Volt Calc: 3 mV
Pack 1, Total temperature sensors: 6
Pack 1, Temp1: 19.5 ℃
Pack 1, Temp2: 18.0 ℃
Pack 1, Temp3: 18.2 ℃
Pack 1, Temp4: 17.0 ℃
Pack 1, Temp5: 28.6 ℃
Pack 1, Temp6: 27.1 ℃
Pack 1, I Pack: 0.0 A
Pack 1, V Pack: 52.476 V
Pack 1, I Remaining Capacity: 43330 mAh
Pack 1, I Full Capacity: 167540 mAh
Pack 1, SOC: 25.86 %
Pack 1, Cycles: 3
Pack 1, Design Capacity: 161000 mAh
Pack 1, SOH: 104.06 %
Script running....
-> Outgoing Data:  b'~250146A60000FD97\r'
<- Incoming data:  b'~25004600400C10ED41723EE4FAEF\r'
SOI:  0x7e
VER:  b'25'
ADR:  b'00'
CID1 (Type):  b'46'
RTN:  b'00'
LENGTH:  b'400C'
 - LCHKSUM:  52
 - LENID:  b'00C'
 - LENID (int):  12
INFO:  b'10ED41723EE4'
CHKSUM:  b'FAEF'
Calc CHKSUM:  FAEF
Pack Remaining Capacity: 43330 mAh
Pack Full Capacity: 167540 mAh
Pack Design Capacity: 161000 mAh
Pack SOC: 25.86 %
Pack SOH: 104.06 %

`

xsherlockpl, 2024-04-05:
Ok , So I'm debugging that further and it looks that my PACE BMS is  having diffrent logic to yours.  I notice that you call 
` success, data = bms_getAnalogData(bms,batNumber=255)`

With batNumber being generic (255) FF
And with in that response you expect there can be multiple packs. 

In my case I have to call that function separately 3 times 

```
 success, data = bms_getAnalogData(bms,batNumber=1)  
 success, data = bms_getAnalogData(bms,batNumber=2)  
 success, data = bms_getAnalogData(bms,batNumber=3)  
```

And I get 3 diffrent battery  datasets each with 1 pack  that correspond to my pack 1 ,2 and 3

Tertiush, 2024-04-05:
Hi, have a look at this configuration option. I'm pretty sure it can help as offsets between vendors appear to differ quite significantly:

force_pack_offset: This is currently available in the development version. This offset is used to force a defined offset between the data read from multiple packs. If you have more than one pack and only the first is read successfully, you can force an offset here to get subsequent packs to read in successfully. Default is 0, multiple of 2 (e.g. 2, 4, 6....) may work. As large as 20 has been used in one instance.

Tertiush, 2024-04-05:
> Ok , So I'm debugging that further and it looks that my PACE BMS is having diffrent logic to yours. I notice that you call ` success, data = bms_getAnalogData(bms,batNumber=255)`
>
> With `batNumber` [...]

255 is for "all", which means their data should be sent in a single reply.

xsherlockpl, 2024-04-05:
Ahhh... so 255 for my BMS does return only first pack data   therefore the force_pack_offset does not have any effect as it fails to find any additional data in response that is not there. I resorted calling bms_getAnalogData  3 times for my packs and rewrote it to return batNumber  in mqtt responses instead of the pack number. Very dirty hack but I need to start my solar system today  Will push that as separate fork as this completely changes the concept of how you access your BMS. coding that right now.

At least it gets me the output I need

```
battery_bms/availability online
battery_bms/pack_1/v_cells/cell_1 3275
battery_bms/pack_1/v_cells/cell_2 3277
battery_bms/pack_1/v_cells/cell_3 3277
battery_bms/pack_1/v_cells/cell_4 3276
battery_bms/pack_1/v_cells/cell_5 3276
battery_bms/pack_1/v_cells/cell_6 3277
battery_bms/pack_1/v_cells/cell_7 3277
battery_bms/pack_1/v_cells/cell_8 3276
battery_bms/pack_1/v_cells/cell_9 3277
battery_bms/pack_1/v_cells/cell_10 3276
battery_bms/pack_1/v_cells/cell_11 3276
battery_bms/pack_1/v_cells/cell_12 3277
battery_bms/pack_1/v_cells/cell_13 3276
battery_bms/pack_1/v_cells/cell_14 3276
battery_bms/pack_1/v_cells/cell_15 3277
battery_bms/pack_1/v_cells/cell_16 3276
battery_bms/pack_1/cells_max_diff_calc 2
battery_bms/pack_1/temps/temp_1 21.1
battery_bms/pack_1/temps/temp_2 20.3
battery_bms/pack_1/temps/temp_3 20.8
battery_bms/pack_1/temps/temp_4 19.8
battery_bms/pack_1/temps/temp_5 24.5
battery_bms/pack_1/temps/temp_6 23.6
battery_bms/pack_1/i_pack 0.0
battery_bms/pack_1/v_pack 52.474
battery_bms/pack_1/i_remain_cap 43230
battery_bms/pack_1/i_full_cap 167540
battery_bms/pack_1/soc 25.8
battery_bms/pack_1/cycles 3
battery_bms/pack_1/i_design_cap 161000
battery_bms/pack_1/soh 104.06
battery_bms/pack_2/v_cells/cell_1 3280
battery_bms/pack_2/v_cells/cell_2 3281
battery_bms/pack_2/v_cells/cell_3 3280
battery_bms/pack_2/v_cells/cell_4 3283
battery_bms/pack_2/v_cells/cell_5 3281
battery_bms/pack_2/v_cells/cell_6 3280
battery_bms/pack_2/v_cells/cell_7 3282
battery_bms/pack_2/v_cells/cell_8 3281
battery_bms/pack_2/v_cells/cell_9 3282
battery_bms/pack_2/v_cells/cell_10 3281
battery_bms/pack_2/v_cells/cell_11 3282
battery_bms/pack_2/v_cells/cell_12 3279
battery_bms/pack_2/v_cells/cell_13 3282
battery_bms/pack_2/v_cells/cell_14 3280
battery_bms/pack_2/v_cells/cell_15 3281
battery_bms/pack_2/v_cells/cell_16 3281
battery_bms/pack_2/cells_max_diff_calc 4
battery_bms/pack_2/temps/temp_1 13.4
battery_bms/pack_2/temps/temp_2 13.6
battery_bms/pack_2/temps/temp_3 12.6
battery_bms/pack_2/temps/temp_4 13.1
battery_bms/pack_2/temps/temp_5 15.7
battery_bms/pack_2/temps/temp_6 15.8
battery_bms/pack_2/i_pack 0.0
battery_bms/pack_2/v_pack 52.496
battery_bms/pack_2/i_remain_cap 45550
battery_bms/pack_2/i_full_cap 167280
battery_bms/pack_2/soc 27.23
battery_bms/pack_2/cycles 2
battery_bms/pack_2/i_design_cap 161000
battery_bms/pack_2/soh 103.9
battery_bms/pack_3/v_cells/cell_1 3269
battery_bms/pack_3/v_cells/cell_2 3270
battery_bms/pack_3/v_cells/cell_3 3269
battery_bms/pack_3/v_cells/cell_4 3268
battery_bms/pack_3/v_cells/cell_5 3267
battery_bms/pack_3/v_cells/cell_6 3268
battery_bms/pack_3/v_cells/cell_7 3268
battery_bms/pack_3/v_cells/cell_8 3268
battery_bms/pack_3/v_cells/cell_9 3267
battery_bms/pack_3/v_cells/cell_10 3268
battery_bms/pack_3/v_cells/cell_11 3267
battery_bms/pack_3/v_cells/cell_12 3268
battery_bms/pack_3/v_cells/cell_13 3270
battery_bms/pack_3/v_cells/cell_14 3268
battery_bms/pack_3/v_cells/cell_15 3269
battery_bms/pack_3/v_cells/cell_16 3266
battery_bms/pack_3/cells_max_diff_calc 4
battery_bms/pack_3/temps/temp_1 13.6
battery_bms/pack_3/temps/temp_2 13.5
battery_bms/pack_3/temps/temp_3 12.8
battery_bms/pack_3/temps/temp_4 12.9
battery_bms/pack_3/temps/temp_5 16.3
battery_bms/pack_3/temps/temp_6 15.5
battery_bms/pack_3/i_pack 0.0
battery_bms/pack_3/v_pack 52.29
battery_bms/pack_3/i_remain_cap 36280
battery_bms/pack_3/i_full_cap 167710
battery_bms/pack_3/soc 21.63
battery_bms/pack_3/cycles 5
battery_bms/pack_3/i_design_cap 161000
battery_bms/pack_3/soh 104.17

```

closed 2024-05-26 by github-actions[bot]

shingsam, 2024-09-01:
> Ahhh... so 255 for my BMS does return only first pack data therefore the force_pack_offset does not have any effect as it fails to find any additional data in response that is not there. I tried calling bms_getAnalogData 3 times for my packs and rewrote it to return batNumber in mqtt responses instead of the pack number. Very dirty hack..
> 
> It is somehow working now, but I had a problem since my packs had wrong address jumpers 0,2,3 i fixed that and it works now with this hack
> 
> ```
> battery_bms/availability online
> battery_bms/pack_1/v_cells/cell_1 3275
> battery_bms/pack_1/v_cells/cell_2 3277
> battery_bms/pack_1/v_cells/cell_3 3277
> battery_bms/pack_1/v_cells/cell_4 3276
> battery_bms/pack_1/v_cells/cell_5 3276
> battery_bms/pack_1/v_cells/cell_6 3277
> battery_bms/pack_1/v_cells/cell_7 3277
> battery_bms/pack_1/v_cells/cell_8 3276
> battery_bms/pack_1/v_cells/cell_9 3277
> battery_bms/pack_1/v_cells/cell_10 3276
> battery_bms/pack_1/v_cells/cell_11 3276
> battery_bms/pack_1/v_cells/cell_12 3277
> battery_bms/pack_1/v_cells/cell_13 3276
> battery_bms/pack_1/v_cells/cell_14 3276
> battery_bms/pack_1/v_cells/cell_15 3277
> battery_bms/pack_1/v_cells/cell_16 3276
> battery_bms/pack_1/cells_max_diff_calc 2
> battery_bms/pack_1/temps/temp_1 21.1
> battery_bms/pack_1/temps/temp_2 20.3
> battery_bms/pack_1/temps/temp_3 20.8
> battery_bms/pack_1/temps/temp_4 19.8
> battery_bms/pack_1/temps/temp_5 24.5
> battery_bms/pack_1/temps/temp_6 23.6
> battery_bms/pack_1/i_pack 0.0
> battery_bms/pack_1/v_pack 52.474
> battery_bms/pack_1/i_remain_cap 43230
> battery_bms/pack_1/i_full_cap 167540
> battery_bms/pack_1/soc 25.8
> battery_bms/pack_1/cycles 3
> battery_bms/pack_1/i_design_cap 161000
> battery_bms/pack_1/soh 104.06
> battery_bms/pack_2/v_cells/cell_1 3280
> battery_bms/pack_2/v_cells/cell_2 3281
> battery_bms/pack_2/v_cells/cell_3 3280
> battery_bms/pack_2/v_cells/cell_4 3283
> battery_bms/pack_2/v_cells/cell_5 3281
> battery_bms/pack_2/v_cells/cell_6 3280
> battery_bms/pack_2/v_cells/cell_7 3282
> battery_bms/pack_2/v_cells/cell_8 3281
> battery_bms/pack_2/v_cells/cell_9 3282
> battery_bms/pack_2/v_cells/cell_10 3281
> battery_bms/pack_2/v_cells/cell_11 3282
> battery_bms/pack_2/v_cells/cell_12 3279
> battery_bms/pack_2/v_cells/cell_13 3282
> battery_bms/pack_2/v_cells/cell_14 3280
> battery_bms/pack_2/v_cells/cell_15 3281
> battery_bms/pack_2/v_cells/cell_16 3281
> battery_bms/pack_2/cells_max_diff_calc 4
> battery_bms/pack_2/temps/temp_1 13.4
> battery_bms/pack_2/temps/temp_2 13.6
> battery_bms/pack_2/temps/temp_3 12.6
> battery_bms/pack_2/temps/temp_4 13.1
> battery_bms/pack_2/temps/temp_5 15.7
> battery_bms/pack_2/temps/temp_6 15.8
> battery_bms/pack_2/i_pack 0.0
> battery_bms/pack_2/v_pack 52.496
> battery_bms/pack_2/i_remain_cap 45550
> battery_bms/pack_2/i_full_cap 167280
> battery_bms/pack_2/soc 27.23
> battery_bms/pack_2/cycles 2
> battery_bms/pack_2/i_design_cap 161000
> battery_bms/pack_2/soh 103.9
> battery_bms/pack_3/v_cells/cell_1 3269
> battery_bms/pack_3/v_cells/cell_2 3270
> battery_bms/pack_3/v_cells/cell_3 3269
> battery_bms/pack_3/v_cells/cell_4 3268
> battery_bms/pack_3/v_cells/cell_5 3267
> battery_bms/pack_3/v_cells/cell_6 3268
> battery_bms/pack_3/v_cells/cell_7 3268
> battery_bms/pack_3/v_cells/cell_8 3268
> battery_bms/pack_3/v_cells/cell_9 3267
> battery_bms/pack_3/v_cells/cell_10 3268
> battery_bms/pack_3/v_cells/cell_11 3267
> battery_bms/pack_3/v_cells/cell_12 3268
> battery_bms/pack_3/v_cells/cell_13 3270
> battery_bms/pack_3/v_cells/cell_14 3268
> battery_bms/pack_3/v_cells/cell_15 3269
> battery_bms/pack_3/v_cells/cell_16 3266
> battery_bms/pack_3/cells_max_diff_calc 4
> battery_bms/pack_3/temps/temp_1 13.6
> battery_bms/pack_3/temps/temp_2 13.5
> battery_bms/pack_3/temps/temp_3 12.8
> battery_bms/pack_3/temps/temp_4 12.9
> battery_bms/pack_3/temps/temp_5 16.3
> battery_bms/pack_3/temps/temp_6 15.5
> battery_bms/pack_3/i_pack 0.0
> battery_bms/pack_3/v_pack 52.29
> battery_bms/pack_3/i_remain_cap 36280
> battery_bms/pack_3/i_full_cap 167710
> battery_bms/pack_3/soc 21.63
> battery_bms/pack_3/cycles 5
> battery_bms/pack_3/i_design_cap 161000
> battery_bms/pack_3/soh 104.17
> ```

Hi, my issue same with u....
Are u success read all pack? Please lets me know.

Thanks