Tertiush/bmspace #39: The command '/bin/sh -c pip3 install -r requirements.txt' returned a non-zero code: 1

tonyp740613, 2024-05-02:
I have been using this addon for more than 12 months and since 30 April it stopped running on it's own and would not start.  I  am running Home Assistant OS 2024.4.4 running on HassOS 12.2

I uninstalled BMS Pace addon 2.2.0 and then attempted to install it again whereupon I received the following output in the supervisor log:

Collecting paho_mqtt==1.6.1
  Downloading paho-mqtt-1.6.1.tar.gz (99 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 99.4/99.4 kB 2.0 MB/s eta 0:00:00
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting PyYAML==6.0
  Downloading PyYAML-6.0.tar.gz (124 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 125.0/125.0 kB 5.7 MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'error'
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [48 lines of output]
      running egg_info
      writing lib/PyYAML.egg-info/PKG-INFO
      writing dependency_links to lib/PyYAML.egg-info/dependency_links.txt
      writing top-level names to lib/PyYAML.egg-info/top_level.txt
      Traceback (most recent call last):
        File "/usr/local/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/usr/local/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/usr/local/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 325, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 295, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 288, in <module>
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/__init__.py", line 104, in setup
          return distutils.core.setup(**attrs)
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 184, in setup
          return run_commands(dist)
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 200, in run_commands
          dist.run_commands()
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/dist.py", line 967, in run_command
          super().run_command(command)
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/command/egg_info.py", line 321, in run
          self.find_sources()
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/command/egg_info.py", line 329, in find_sources
          mm.run()
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/command/egg_info.py", line 550, in run
          self.add_defaults()
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/command/egg_info.py", line 588, in add_defaults
          sdist.add_defaults(self)
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/command/sdist.py", line 102, in add_defaults
          super().add_defaults()
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/_distutils/command/sdist.py", line 250, in add_defaults
          self._add_defaults_ext()
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/_distutils/command/sdist.py", line 335, in _add_defaults_ext
          self.filelist.extend(build_ext.get_source_files())
        File "<string>", line 204, in get_source_files
        File "/tmp/pip-build-env-4tk9q44q/overlay/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 107, in __getattr__
          raise AttributeError(attr)
      AttributeError: cython_sources
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

[notice] A new release of pip is available: 23.0.1 -> 24.0
[notice] To update, run: pip install --upgrade pip

 ---> Removed intermediate container 3868d2cd0e88

tonyp740613, 2024-05-02:
There is another entry in the Home Assistant Core log:

Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:135
integration: Home Assistant Supervisor ([documentation](https://www.home-assistant.io/integrations/hassio), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+hassio%22))
First occurred: 9:44:23 AM (1 occurrences)
Last logged: 9:44:23 AM

Failed to to call /addons/local_bms_pace/install - The command '/bin/sh -c pip3 install -r requirements.txt' returned a non-zero code: 1

Tertiush, 2024-05-02:
That's weird, I just uninstalled mine and reinstalling worked fine, see below. You could try the development version, which has a few additional features. However the dependencies are the same....

`
2024-05-02 10:48:53.141 INFO (SyncWorker_4) [supervisor.docker.manager] Removing image 1f71505a/amd64-addon-bms_pace with latest
2024-05-02 10:48:53.142 INFO (SyncWorker_4) [supervisor.docker.manager] Removing image 1f71505a/amd64-addon-bms_pace with 2.2.0
2024-05-02 10:48:53.192 INFO (MainThread) [supervisor.addons.addon] Removing add-on data folder /data/addons/data/1f71505a_bms_pace
2024-05-02 10:48:53.205 INFO (MainThread) [supervisor.addons.manager] Add-on '1f71505a_bms_pace' successfully removed
2024-05-02 10:49:16.023 INFO (MainThread) [supervisor.addons.addon] No 1f71505a_bms_pace addon Docker image 1f71505a/amd64-addon-bms_pace found
2024-05-02 10:49:16.029 INFO (MainThread) [supervisor.docker.addon] Starting build for 1f71505a/amd64-addon-bms_pace:2.2.0
2024-05-02 10:49:24.970 INFO (MainThread) [supervisor.docker.addon] Build 1f71505a/amd64-addon-bms_pace:2.2.0 done
2024-05-02 10:49:24.978 INFO (MainThread) [supervisor.addons.addon] Creating Home Assistant add-on data folder /data/addons/data/1f71505a_bms_pace
2024-05-02 10:49:24.984 INFO (MainThread) [supervisor.docker.addon] Starting build for 1f71505a/amd64-addon-bms_pace:2.2.0
2024-05-02 10:49:27.267 INFO (MainThread) [supervisor.docker.addon] Build 1f71505a/amd64-addon-bms_pace:2.2.0 done
2024-05-02 10:49:27.267 INFO (MainThread) [supervisor.addons.manager] Add-on '1f71505a_bms_pace' successfully installed
`

tonyp740613, 2024-05-02:
I removed the addon from my Home Assistant and added the repository again.  I then chose the DEV version which installed perfectly.  The addon is working again.